jgaeddert/liquid-dsp #171: Implementation of Kaiser window function

clydeclements, 2019-08-15:
The documentation in the code for the `kaiser` function in `windows.c` states that parameter `_N` is the window length. The source for that function includes
```
    float t = (float)_n - (float)(_N-1)/2 + _mu;
    float r = 2.0f*t/(float)(_N);
    float a = liquid_besseli0f(_beta*sqrtf(1-r*r));
    float b = liquid_besseli0f(_beta);
    return a / b;
```
Parameter `_N` is used in a denominator above, but I believe the customary definition is to use `_N - 1` such as in [MATLAB](https://www.mathworks.com/help/signal/ref/kaiser.html) and [SciPy](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.windows.kaiser.html). So, shouldn't the second line in the code above be as follows?
```
    float r = 2.0f*t/(float)(_N-1);
```

jgaeddert, 2019-08-17:
You are correct in that it should be _N-1; however I'm getting some errors with the autotest checks so I'll need to track those down before I can commit the fix. Thanks!

jgaeddert, 2019-09-14:
fixed in dead3b5a5abf3c6244ce9718c81b026f77748f48

closed 2019-09-14 by jgaeddert