jgaeddert/liquid-dsp #173: SNR option in qdetector example seems incorrect

garbagetrash, 2019-09-06:
This is only in the example code, so I don't think it's necessarily a high priority.

I was messing with the qdetector example provided and was surprised to see it working quite well at even -10 dB SNR (with otherwise default settings.)  I ran the plotting code the example spits out after doing a:

``` sh
$ qdetector_cccf_example -S-10
```

and got this plot:

[image]

Which is pretty obviously not -10 dB SNR burst.  Looking at the example code itself it appears that the noise power is referenced to 1.0 versus the actual signal power output from the filter, which doesn't really agree with the definition of SNR.

Relevant code:

``` C
    // get pointer to sequence and generate full sequence
    float complex * v = (float complex*) qdetector_cccf_get_sequence(q);
    unsigned int filter_delay = 15;
    firfilt_crcf filter = firfilt_crcf_create_kaiser(2*filter_delay+1, 0.4f, 60.0f, -tau);
    float        nstd        = powf(10.0f, -SNRdB/20.0f);
    for (i=0; i<num_samples; i++) {
        // add delay
        firfilt_crcf_push(filter, i < seq_len ? v[i] : 0);
        firfilt_crcf_execute(filter, &y[i]);

        // channel gain
        y[i] *= gamma;

        // carrier offset
        y[i] *= cexpf(_Complex_I*(dphi*i + phi));
        
        // noise
        y[i] += nstd*(randnf() + _Complex_I*randnf())*M_SQRT1_2;   // <--- This needs to be scaled by the actual signal power
    }
    firfilt_crcf_destroy(filter);
```

I'm not sure if this error exists in the other examples or not as I only just found it, but you might want to review and correct it when you get the chance.

Liquid-DSP is a fantastic library, though, thank you for your extensive work on this over the years!

garbagetrash, 2019-09-07:
Looking closer at the code I realize the channel gain is just a hardcoded value (10.0).  I propose removing channel gain entirely so that the signal power is also unitary, and then the SNR argument ought to be correct as they both would be referenced to 1.0.

jgaeddert, 2019-09-14:
Good catch! I committed a fix before I realized you had a PR, so I went ahead and merged your changes in. 👍

closed 2019-09-14 by jgaeddert