jgaeddert/liquid-dsp #176: FSK Framesync doesn't reset state after invalid header

alog243, 2019-09-26:
Self-explanatory title.

fskframesync.c line 577 doesn't have a `fskframesync_reset(_q);` like other framing schemes.
This means if you ever receive in invalid header the framesync state is stuck until it corrupts the heap and crashes because of line 521:

> `_q->header_sym[_q->symbol_counter++] = sym;`

The only check against the symbol counter is an == check, instead of a >= check (and it happens after you write the heap so that doesn't stop it from corrupting anyway).

There's no way to actually recover from this state without the framesync reset.

jgaeddert, 2019-09-27:
Should be resolved with 744fb4d60fac4771a4cc32946023d1305f13a219. Please verify.

alog243, 2019-09-27:
Yup, it works now. I didn't look at all of the other framesync objects so I don't know if the reset is missing anywhere else; I was just working with the FSK frames when I noticed it.

closed 2019-09-27 by alog243

jgaeddert, 2019-09-27:
It's a fair point. This is one of those areas that having more unit tests (e.g. `make check`) for these frame synchronizers would be useful.

jgaeddert, 2019-09-27:
(btw, thanks!)