jgaeddert/liquid-dsp #189: Violation of section 7.1.3 of the C standard (ISO 9899:1999)

TehWan, 2020-02-04:
The source code and headers use variables named with a leading underscore followed by a capital letter. This is reserved for implementation purposes and is not guaranteed to work on every machine. This issue is the main reason I am unable to use liquid-dsp on Windows. The `_X` variable actually resolves to the value `64` on my machine, which is not a valid name.

See https://stackoverflow.com/a/1449301/248188

TehWan, 2020-02-04:
I committed a fix in my fork. See following commit: https://github.com/TehWan/liquid-dsp/commit/c617e4716202a4d6de37fc6caaf2088be3ee2253

jgaeddert, 2020-03-01:
Yikes. I wasn't aware of this. Shouldn't be hard to fix, but would create massive diffs. I had tried to reserve variable names for function arguments with leading underscores to help with consistency. This seems like one of those things where convention starts to influence the standard.

TehWan, 2020-03-01:
The issue is not with the variable names in the C files, but with the fact you use macros to define your interface in your header file. Because of macro expansion, names with leading underscores might get expanded into their value if defined. No need to modify the C files for that, only the headers, which technically don't require names for the variables, only types.

jgaeddert, 2020-03-01:
Well, I do use macros to expand the source files as well to emulate templated types.