jgaeddert/liquid-dsp #190: DSSS, longer p/n sequence

vladkoliesnik, 2020-03-25:
Hi, DSSS implementation is great, but what about using more longer p/n sequence than 2^7? Is it possible with this library? I want to see difference between 127, 511, 1023 and longer p/n sequence. I am already done test dsssgensync/gen with LimeSDR and want to move forward. 

So, its possible without ruining your lib? =)
Thank you!

brian-armstrong, 2020-12-15:
This is likely possible but not just by changing a single constant. The polynomial is controlled here https://github.com/jgaeddert/liquid-dsp/blob/master/src/framing/src/dsssframegen.c#L129

You would have to change both dsssframegen.c and dsssframesync.c and then change all those 64s to be `2^(n-1)`s. It'd take a bit of patience but should be doable.

brian-armstrong, 2020-12-15:
Also, the preamble of the frame uses a different polynomial so that it can be distinctly detected separately from the rest of the payload. You would likely want to change it as well. https://github.com/jgaeddert/liquid-dsp/blob/master/src/framing/src/dsssframegen.c#L121