jgaeddert/liquid-dsp #191: Question regarding firhilb

mryndzionek, 2020-04-10:
Hello Joseph,
I have two processing flows.
One is:
```sh
ccf32samples -> freqdem_demodulate -> iirfilt_rrrf_execute ->
firdecim_rrrf_execute -> wavfile wavfile (float samples)
```
and the second operates on complex IQ signal, i.e. has `firhilbf_decim_execute` and `firhilbf_interp_execute` inserted at beginning and end:

```sh
ccf32samples -> freqdem_demodulate -> firhilbf_decim_execute ->
iirfilt_crcf_execute -> firdecim_crcf_execute -> firhilbf_interp_execute ->
wavfile (float samples)
```
I was expecting the result of the two processing step to be the same, or at least similar, but the second flow introduces nasty (high frequency) sound artifacts. Am I misunderstanding something?
Should I something different than `firhilbf` for converting CF32 <-> Float signals?

fsheikh, 2020-04-10:
Although I am not sure about modulated `ccf32samples` you may have, imho after `freqdem_demouldate` you already have the passband signal extracted and then the equivalent low pass signal should be there by using `firhilbf_decim_execute`. In order to recover I/Q data streams you can just ask for real and imaginary components from the complex out of `firhilbf_decim_execute`

mryndzionek, 2020-04-10:
I'm not sure what you mean. I don't have a problem with extracting Re Im components from the IQ stream, but converting IQ (complex float) to real (float) signal.

My original question is probably equivalent to the two below:
 - are `firhilbf_decim_execute` and `firhilbf_interp_execute` inverses of each other?
 - do the `iirfilt_*` and `firdecim_*` filters in their `rrrf` versions work exactly the same as `crcf` versions?

fsheikh, 2020-04-10:
I guess what I did not understand was what do you want to achieve by converting an IQ sample to a single *real* value (you are throwing away information). Anyway if you plot the results of `examples/firhilb_example.c` you will notice that the `firhilb_decim` and `firhilb_interp` are not strict inverses of each other (meaning point-wise samples comparison would not work) but spectral properties should remain the same.

mryndzionek, 2020-04-10:
Yeah, I think this makes sense, so probably my strange result/problem was caused by something else.

jgaeddert, 2020-04-11:
Interesting. My guess is that there is some allowing that’s happening in the second flow. I could see this happening if the decimation tightens the spectrum up around the signal to the point where the hilbert transform can’t isolate. Hard to tell without seeing the numbers but in any case the first flow is definitely best. You don’t lose anything by operating on real samples.

jgaeddert, 2020-04-11:
*aliasing (not "allowing"). DYAC!

mryndzionek, 2020-04-11:
Thanks for the reply. I'll use the first flow for now. Will investigate further the next time.

mryndzionek, 2020-04-11:
One other question regarding the PFB channelizer. Looking [here](https://github.com/jgaeddert/liquid-dsp/blob/master/examples/firpfbch_crcf_analysis_example.c#L69)
and [here](https://github.com/jgaeddert/liquid-dsp/blob/master/examples/firpfbch_crcf_analysis_example.c#L103) it seems that in order to have the output in separate arrays the `y` array would need a transpose, correct?

closed 2020-04-12 by mryndzionek