jgaeddert/liquid-dsp #192: Question regarding `firpfbch_crcf`

mryndzionek, 2020-04-12:
I'm testing the `firpfbch_crcf` channelizer filterbank and I'm getting somewhat strange results. In below screenshot:
[image]

On the left is the full spectrum my SDR is tuned to: 92MHz center frequency and 200kHz bandwidth. On the right is the the `firpfbch_crcf` channelizer output when run on this signal - 4 channels, 50kHz each. Two things to notice:
 - it's asymmetrical/shifted - the top channel
 - the order of the signals is strange. I've adjusted them, the filenames indicate the real order (o1.cf32, o2.cf32, o3.cf32, o4.cf32) - the channels 'go up', but o4 is at the bottom

Am I doing something wrong here?

jgaeddert, 2020-04-12:
The output of the channelized follows the same convention as an fft; that is, the DC component is at index 0 and it walks its way up from there. But remember that for discrete-sampled signals, the spectrum wraps around at Fs/2, so you'll need to run an FFT shift if you want the DC component centered.

Put another way, if you have M=4 channels, the output of the channelizer should be at indices 0, 1, 2, and 3 where
  * 0: f=0 (DC)
  * 1: f=Fs/4
  * 2: f=+Fs/2 = -Fs/2
  * 3: f = 3*Fs/4 = -Fs/4

mryndzionek, 2020-04-12:
Thanks for explanation. After some investigation I ended up doing the same as the guys [here](https://github.com/suttonpd/iris_modules/blob/master/components/gpp/phy/PfbChannelizer/PfbChannelizerComponent.cpp#L108) and [here](https://github.com/suttonpd/iris_modules/blob/master/components/gpp/phy/PfbChannelizer/PfbChannelizerComponent.cpp#L140).

closed 2020-05-18 by mryndzionek