jgaeddert/liquid-dsp #195: Feature request : symstream : add fractional sample delay to constructor of symstream

pfeatherstone, 2020-05-14:
Currently the sample delay is hard coded to 0.
It would be nice if that was adjustable.
Thanks

jgaeddert, 2020-05-17:
Are you referring to a fractional sample delay? I decided not to include that as to prevent making the create() method have too many arguments. If you need to include a fractional sample delay, consider adding a filter on its output:

```C
float delay = 0.216; // fractional sample delay
firfilt_crcf delay_filter = firfilt_crcf_create_kaiser(21, 0.45, 60.0, delay);
```

pfeatherstone, 2020-09-09:
@jgaeddert Can you apply a fractional sample delay by applying phase shifts to the FFT?
So:
1. do a forward FFT
2. apply nco object with non-zero 'phase'
3. do a backward FFT

jgaeddert, 2020-09-09:
That is certainly possible, and I started development in a branch to test that a while ago for transformations on FIR filter designs. The only consideration is that the fractional sample delay will be circular, so you will need to account for this in the output signal.

pfeatherstone, 2020-09-09:
How about if you pad the input with zeros before the FFT?

pfeatherstone, 2020-09-09:
Is that an artifact if the FFT ?

jgaeddert, 2020-09-10:
It is indeed an artifact of the FFT. Adding with zeros will account for the delay but will extend the signal duration slightly.