jgaeddert/liquid-dsp #196: liquid_firdes_kaiser : _mu is in range [-0.5,0.5] why not [-1.0,1.0]?

pfeatherstone, 2020-05-14:
The other root-nyquist filters allow the fractional time delay to be in range [-1.0,1.0]

fsheikh, 2020-05-15:
I think its the other way around, since the fractional delay is a fraction of sampling period. Interestingly some of the Nyquist filters don't even use the fractional time delay parameter. I tried to address this in #198 Lets see if @jgaeddert agrees.

pfeatherstone, 2020-05-16:
So if you want a delay of 1 sample, then you need mu to be either -1 or 1. In my case that's very useful. So i would bound mu consistently in range [-1,1]

fsheikh, 2020-05-16:
IIUC, in [firdes_rcos](https://github.com/jgaeddert/liquid-dsp/blob/master/src/filter/src/rcos.c#L39) you can get a delay of 1 sample with `_k=1 _m=1 _dt=0` and in [firdes_kaiser](https://github.com/jgaeddert/liquid-dsp/blob/master/src/filter/src/firdes.c#L244) with `_n=3 _dt=0`
Am I missing something?

pfeatherstone, 2020-05-16:
Ok how about a delay of 0.9 samples.

jgaeddert, 2020-05-17:
Good points, all! A little background: the square-root raised-cosine filter (while despite being a terrible filter) does have a nice closed-form solution. Better filters such as a windowed Kaiser are a little trickier to generate with a fractional sample offset: `h(t) = sinc(t) w(t)` While it is pretty straightforward to specify a timing offset for the `sinc` function, it isn't so much for the window. So the simple way is to just apply a slight time offset to the `sinc` but ignore the delay in the window. Unfortunately this actually distorts the group delay of the filter because the window no longer properly aligns with the `sinc`.

This is all to say that as you start specifying larger delays this way, the filter response gets worse; both in terms of spectral response as well as group delay.

So that's why it's better to keep the fractional delay to a minimum. Technically, any delay outside of [-0.5,+0.5] or [0,1] can be achieved by stuffing (or slipping) a sample in the filter buffer.

I do agree that it should be consistent across filter designs. And there are ways to impart a fractional sample delay to any filter, regardless of being able to represent it in closed form (essentially phase-shifting the FFT of the impulse response). I'm certainly open to suggestions!

jgaeddert, 2020-05-17:
I threw together a quick sandbox program to demonstrate this with 317ff5859631f841009d5d481df5c8c0bdaded5e

Running it should give the following output. Notice that the timing error gets larger with `dt`. I think the best way to move forward would be to implement a way to perform fractional interpolation for any filter in a generic way, and not rely on closed-form solutions to filter designs.
```
# fractional sample offset: specified vs. actual
#
#    specified       actual        error
  -2.000000000 -1.260962963 -0.739037037
  -1.875000000 -1.177994251 -0.697005749
  -1.750000000 -1.095870018 -0.654129982
  -1.625000000 -1.014517784 -0.610482216
  -1.500000000 -0.933875084 -0.566124916
  -1.375000000 -0.853875637 -0.521124363
  -1.250000000 -0.774457932 -0.475542068
  -1.125000000 -0.695559978 -0.429440022
  -1.000000000 -0.617131233 -0.382868767
  -0.875000000 -0.539108753 -0.335891247
  -0.750000000 -0.461443424 -0.288556576
  -0.625000000 -0.384080887 -0.240919113
  -0.500000000 -0.306964397 -0.193035603
  -0.375000000 -0.230050564 -0.144949436
  -0.250000000 -0.153283596 -0.096716404
  -0.125000000 -0.076617718 -0.048382282
   0.000000000  0.000000954 -0.000000954
   0.125000000  0.076616287  0.048383713
   0.250000000  0.153284073  0.096715927
   0.375000000  0.230051041  0.144948959
   0.500000000  0.306965828  0.193034172
   0.625000000  0.384077072  0.240922928
   0.750000000  0.461441040  0.288558960
   0.875000000  0.539108276  0.335891724
   1.000000000  0.617130280  0.382869720
   1.125000000  0.695563316  0.429436684
   1.250000000  0.774456024  0.475543976
   1.375000000  0.853876114  0.521123886
   1.500000000  0.933874130  0.566125870
   1.625000000  1.014519691  0.610480309
   1.750000000  1.095870018  0.654129982
   1.875000000  1.177993774  0.697006226
   2.000000000  1.260963440  0.739036560
```

fsheikh, 2020-05-17:
> perform fractional interpolation for any filter in a generic way,

AFAIU, generic fractional interpolation [method](https://github.com/jgaeddert/liquid-dsp/blob/master/src/filter/src/firfarrow.c#L193) is also available. IMHO a minor problem is consistency through `firdes` objects which can be easily achieved via #198. Please consider if possible