jgaeddert/liquid-dsp #197: msourcecf wideband noise on signals issue

AGS-Nelson, 2020-05-14:
I've been working with the library and noticed when I use msource to have files playback, noise, tones, etc all in one, I can only ever get at most about 60 dB of SNR.  I attached a screenshot of what I am talking about.  Based on my testing it seems like some sort of floating point and numerical error that introduced for specific frequency offsets.  If I place a tone at a multiple of sample rate divided by 1024, I get a nice filter effect, otherwise I get a "skirt" like seen on the noise signal.  The noise signals always have this "skirt" effect, even when generated in a user callback.  This was seen in the msourcecf example if a sufficiently low gain wideband noise signal was used as well.  The FIR filters, even with floating point error, should behave much better.
[image]

jgaeddert, 2020-05-17:
Consider setting the "bandwidth" of the tone to zero in the msource object:

```C
// create msource object
msourcecf gen = msourcecf_create(1200, 4, 60.);

// add signals     (gen, fc,         bw,  gain)
msourcecf_add_tone (gen, 300./1200., 0.0,    0); // filtered tone
msourcecf_add_tone (gen,-300./1200., 0.5,  -25); // will show skirt
```

[image]

AGS-Nelson, 2020-05-27:
I notice if I have a tone offset in frequency by 1/1024 of the normalized bandwidth, I still get a fairly clean tone even when instantaneous frequency measurements are made (but still lousy noise rolloff).  First thing that comes to mind is this may be related to the NCO using a sine/cosine look up table of 1024 elements, and a rather interesting calculation method using float bit manipulation.  I am not sure if this is the reason for it, but have you ran into a similar issue before?  This behavior is shown in my original plot in this issue.  I did notice that when not at 1/1024 increments of center frequency that there was a skirt unless I used the PSD calculation including the initial FIR tap wind-up.  In theory, it should be possible to have clean tones when looking over many samples, without looking at all of the generated data.

jgaeddert, 2020-08-27:
Needs revisiting

JayKickliter, 2020-08-27:
> may be related to the NCO using a sine/cosine look up table of 1024 elements

I've noticed that the NCO does produce a lot of (phase?) noise.