jgaeddert/liquid-dsp #199: iirfilt_crcf_execute: backward filtering crashes when using uint64_t counter

Cuda-Chen, 2020-05-19:
Due to certain reasons, I want to do backward filtering to my signal.

The type signal `data` is double, and output `filterResult` is float complex.
The loop counter type is uint64_t.
When using as follows, I receive `SIGSEGV` on iirfilt_crcf_execute():
```
uint64_t i;
for (i = n - 1; i >= 0; i--)
{
  iirfilt_crcf_execute (q, data[i], &filterResult[i]); // crash here
}
```
Strangely, the program exits normally when setting loop counter type to int:
```
int i;
for (i = n - 1; i >= 0; i--)
{
  iirfilt_crcf_execute (q, data[i], &filterResult[i]); // no any problem
}
```

You can view my code to reproduce such issue here: https://github.com/Cuda-Chen/ms2filter/blob/7bdc01ee70879f7b41654ce895390549ed28ea3b/src/bandpass_filter.c#L73

PS. If I do forward filtering with uint64_t counter, the program ends with no error
```
uint64_t i;
for(i = 0; i < n; i++)
{
  iirfilt_crcf_execute (q, data[i], &filterResult[i]); // still no any problem
}
```

JayKickliter, 2020-05-19:
The problem is that uint64_t is always i >= 0. When the counter is already 0 And is decremented it wraps around to UINT64_MAX which is certainly out of bounds for your array.

JayKickliter, 2020-05-19:
TLDR: always use a signed value for loop counters if you don’t have a reason not to

JayKickliter, 2020-05-19:
I feel like we’re dealing with the [XY](https://en.m.wikipedia.org/wiki/XY_problem). You are bringing up three orthogonal points: filter direction, sample type, and loop counter type. Is your question really about the loop counter or something else?

jgaeddert, 2020-05-19:
Agree with @JayKickliter; Your program crashing because of the loop counter, but the results won't make sense anyway because you're casting a `double *` as a `float *`. You'll need to do the conversion first to `float` before running.

closed 2020-05-19 by jgaeddert

Cuda-Chen, 2020-05-20:
Hi @JayKickliter @jgaeddert , thanks for your help!