jgaeddert/liquid-dsp #200: Compile for intel and Amd (question)

alphafox02, 2020-05-26:
Is there a flag or option I can add when compiling that’ll allow the installed liquid-dsp to run in both intel and amd? I’ve not figured out the exact issue and I’m sure this is a weird request, but if I compile liquid-dsp as it states on your GitHub page to use alongside cubicsdr it works fine on my intel machine. If I take out the hard drive and stick it in a Amd based machine, start up cubicsdr I get failures with liquid-dsp. I know that’s not a lot of detail, but I will say if I go into the liquid-dsp folder make uninstall/clean and rerun the instal steps everything is then fine with it. 

I’m assuming this has something to do with me first compiling on intel and then moving to amd.

jgaeddert, 2020-06-06:
I don't think so. While AMD and Intel share the X86_64 architecture, running a program (or linking to a library) compiled for a different system is going to have unexpected results, and will most certainly not work. You'd want to either cross-compile for the target system, or just build natively.

closed 2020-06-06 by jgaeddert

alphafox02, 2020-06-06:
Hi. I found that compiling as directions state on the github page results as I mentioned above. What I end up having to do in the Amd machine is go back in the liquid-dsp source folder and run make uninstall and or make clean followed by all the steps to compile it from scratch. Everything is then fine on the Amd machine. If I understand what you are saying the standard steps on the liquid dsp git page will not allow me to simply move from Intel to Amd without rebuilding it.