jgaeddert/liquid-dsp #201: MATLAB Plotting Not Working in resamp_crcf_example.c

AGS-Nelson, 2020-05-27:
Seemingly simple issue, the syntax used for plotting is not working and gives errors in R2019a MATLAB.  
Examples:
```
Error using plot
Invalid data argument.

Error in resamp_crcf_example (line 887)
  plot(tx,real(x),'-s','Color',[0.5 0.5 0.5],'MarkerSize',1,...
```

```
Error using plot
Invalid data argument.

Error in resamp_crcf_example (line 894)
  plot(tx,imag(x),'-s','Color',[0.5 0.5 0.5],'MarkerSize',1,...
```

```
Error using plot
Vectors must be the same length.

Error in resamp_crcf_example (line 913)
  plot(fx,X,'Color',[0.5 0.5 0.5],fy,Y,'LineWidth',2);
```
I just removed the extra plotting flags and added a '-o' option to the resampler output to make it look decent, not sure if there are issues in other examples too.
[image]

jgaeddert, 2020-06-06:
Yes, my apologies. I don't really use MATLAB, but rather [octave](https://www.gnu.org/software/octave/) which has a few differences in plotting commands.

closed 2020-06-06 by jgaeddert

AGS-Nelson, 2020-06-08:
Ah ok, I was wondering if that was the case.  I will keep that in mind, should be possible to have the commands work the same in MATLAB and Octave with some tweaks.