jgaeddert/liquid-dsp #202: Msource improper resampling

AGS-Nelson, 2020-05-27:
I had generated and verified files with a single 100 Hz tone at various sample rates (1 kHz sample rate, 10 kHz, etc up to 100 MHz sample rate).  I loaded these files into memory and sampling them using the user source in the msource object.  I notice when I set the user source bandwidth equal to this original sample rate divided by my overall sample rate for the output, I correctly get the files re-sampled so that they all look to be the same frequency.  That is until my overall normalized bandwidth gets too small (I start noticing issues around ~<0.001 bandwidth for the user source).  As the bandwidth for the output data keeps going up (effectively making the BW of each user source smaller) I notice the timing distortion becomes worse and eventually it just doesn't resample at all and the frequencies are clearly different.
I recreated the issue with minimal changes to the code structure in msource, I am attaching the file sources I used and the example code which displays the issue nicely.  Anyone can by eye tell that the frequency is off significantly by counting the peaks/troughs.  Looking at resampler rate vs actual resampler rate doesn't seem to account for the issue alone.
The person verifying this may need to modify the file path macros depending on where they put these files.  If you run the example for the different USE_FILE macro values (1-3) and run the resulting MATLAB script you should see the same plots.  I am also including the plots here for reference.  A warning that MATLAB is extremely slow using such large source files (48000 complex data points in it since I wanted to not disturb the code structure as much as possible).  I also notice it seems like when this happens there are distortions going beyond the initial 9600 FIR filter taps in the samples, seen when comparing the sharp cutoff at higher sample rates at 9600 samples.
[Test_Data.zip](https://github.com/jgaeddert/liquid-dsp/files/4691047/Test_Data.zip)
[image]
[image]
[image]

AGS-Nelson, 2020-05-27:
Seems to be related to the P value not being allowed to go below 2 (required for the fir polyphase filter bank channelizer #2).  It may require resampling beforehand before being able to handle such narrowband signal inputs.

AGS-Nelson, 2020-06-02:
It seems there are skews in the frequency output even before the 1/600 bandwidth lower limit, I've had the best results sticking to signals of bandwidth above 0.05.

jgaeddert, 2020-06-06:
I guess I'm not exactly sure what the issue is. The transient visible at the beginning of the burst is directly related to the bandwidth of the filter applied to the signal (essentially the bandwidth parameter given to the msource object). Is part of the issue the delay that's imposed by the channelizer?

AGS-Nelson, 2020-06-08:
In this case the transient is an issue but not unexpected, rather the frequency being incorrect by a significant amount is the most pressing concern.  Something about the bandwidth not being compensated for correctly at small bandwidths.  If I resample a waveform before putting it to msource, and limit msource bandwidth to not go below 0.05, I get fairly good results.

jgaeddert, 2020-06-18:
Can you provide a simple example that reproduces this?

AGS-Nelson, 2020-06-18:
Woops I thought I had included this the example source code in that zip, guess not...  Here is the code I ran.  Combining the MSource with an extra arbitrary resampler beforehand has given me fairly good results, but then needs an extra stage for very narrowband files.
[test_code.zip](https://github.com/jgaeddert/liquid-dsp/files/4801226/test_code.zip)

jgaeddert, 2020-06-18:
Ok, I'll take a look at it. I threw something together quickly, and I'm definitely seeing some odd behavior.

https://gist.github.com/jgaeddert/7215bdc268e0d32e24fed0b6f7f26e67