jgaeddert/liquid-dsp #203: gmskdem - is it possible to compensate drift?

mryndzionek, 2020-06-07:
As in the topic. It seems that GMSK requires perfect sample timing - I'm seeing the symbols 'drifting' more and more with time. Would it be possible to somehow do active frequency drift compensation.

JayKickliter, 2020-06-08:
Slightly off-topic, but I tried using Liquid to demodulate real-world streaming GMSK with no success. The best tool I could find is the newish [symbol_sync][1] block from GNU Radio. The math is above my head, but it would be great to see someone port the synchronization schemes to Liquid.

[1]: https://www.youtube.com/watch?v=uMEfx_l5Oxk

jgaeddert, 2020-06-09:
Absolutely! One of the first [DSP algorithms I wrote](https://github.com/jgaeddert/liquid-dsp/commits/master/src/filter/src/symsync.c) for liquid was a symbol-timing recovery component using polyphase filter banks. It was originally designed for linear modulation schemes (like QPSK) but it works quite well for GMSK as well. There is a basic description for how to use it [here](https://liquidsdr.org/doc/symsync/).

If you also want to incorporate automatic gain control, carrier frequency/phase correction, timing recovery, and equalization, take a look at the [symtrack](https://liquidsdr.org/doc/symtrack/) object which integrates all of these together while ensuring stability across the various loops.

mryndzionek, 2020-06-13:
I'm definitely going to make more tests with sym_sync and sym_track.
What I've tried so far:

`resampled IQ,FSK, ~5samples/symbol -> gmskdem (k=5, m=1)`

This sort of works. I can decode the data, but is slowly goes out of sync, more errors
appear and then it goes back in sync and this happens periodically throught all the data.

`resampled IQ,FSK, ~5samples/symbol -> symsync(k=5) -> gmskdem (k=5, m=1)`

This doesn't work at all. Maybe I'm setting 'samples per symbol' wrong?
Maybe I should use more samples/symbol on input, like 30. Then output
of symsync with k=5 would be 6, as it's a decimating block, right?

I was also thinking about running 'freqdem' on the input signal to get 
something resembling a square wave and run symsync/symtrack on this and then try
to decode the bitstream.

Any thoughts/advice is appreciated.

jgaeddert, 2020-06-13:
Yes, the gmsk demodulator alone does not perform symbol timing recovery. Your second approach of applying the symsync before `gmskdem` is ideal and should provide the best performance. I'll look into why that's not working for you.

But to help you move forward with your problem, I implemented a solution that you suggested last: run the FM demod *before* symbol timing recovery. See d1fd949 which adds the `gmsk_tracking_example.c` program. It uses 4 samples per symbol, resamples at a rate P/Q=203/200 which is 1.5% increased rate. Running it should provide an Octave script that gives a plot like the following:

[image]

I tried adjusting `k=5` in the example, but it did take a little longer to converge (needed to increase the simulation duration a bit); alternatively increasing the `symync` loop filter bandwidth from `0.02` to `0.05` did the trick.

Let me know if you have any questions with this, and I hope this was helpful.

mryndzionek, 2020-06-13:
What exactly is the purpose of the matched filter? I've made some test. Here is the plot of my FM demodulated data:
[image]
After applying `symsync`:
[image]
With the filter in between it's worse:
[image]

jgaeddert, 2020-06-14:
The matched filter is intended to remove inter-symbol interference (ISI) from the transmitted signal. This is especially prevalent with GMSK which specifically adds ISI to reduce the signal's occupied bandwidth.

I will also point out that the simulation example I added does not include noise. You should probably apply a low-pass filter before the FM demod to remove any out-of-band noise. This noise would get amplified with the FM demod, so it's a good idea to remove it first.

I can't speak to why the matched filter would make your signal worse, unless it isn't truly matched to the transmitted signal's GMSK filter. In the example, I used a bandwidth-time factor (`beta`) value of 0.3. This might need to be adjusted to match the transmitted signal you're working with.

And as a last point, you can always add a dynamic equalizer *after* the symsync object to help remove any ISI that isn't removed by the matched filter.

To summarize, try the following:

  1. Resample to ~4 samples per symbol (keeps processing reasonably low)
  1. Apply a low-pass filter (e.g. 3rd-order butterworth) to remove out-of-band noise
  1. Run the FM demodulator, setting the modulation index such that the output has an RMS value of about 1.0
  1. Run a low-pass `firfilt` filter (ideally a matched filter to remove ISI) with a frequency cut-off of about 0.125 or maybe a little higher
  1. Run through the symbol synchronizer with an output rate of 2 samples/symbol
  1. Run the result through an `eqlms_rrrf` object with a decision-directed output (target +1/-1 values)

I understand this seems fairly complicated, but it does have many advantages in that the maximum amount of noise is always filtered out at each stage.

mryndzionek, 2020-06-14:
1. I've tried 4 samples per symbol - not much difference
2. Applied low-pass filter, but results still worse than without it. I'm applying the filter after initial resampling to 4 samples per symbol. Maybe I should apply the filter before resampling?
3. Made some attempts, but no significant improvement. My original waveform after demodulation is something like [this](https://github.com/mryndzionek/composable-sdr/blob/devel/images/ex6_2.png).
5. This is what I'm currently doing.

Will experiment more with (4) and (6) later this week. I've made a more detailed write-up of what I'm doing [here](https://github.com/mryndzionek/composable-sdr/tree/devel#example-6). The resulta after applying symbol synchronizer are good. Maybe it's the best that can be done with a signal like this?

Thanks for all the help!

closed 2021-03-13 by mryndzionek

xerpi, 2024-07-26:
Sorry for commenting on this closed issue, but is it possible to use `symtrack` for GMSK?
What would be the ideal filter type and modulation scheme to pass to `symtrack_cccf_create`?
Since there's no `LIQUID_MODEM_GMSK` (I guess since it's not linear), and https://liquidsdr.org/doc/symtrack/ says "modulation scheme (can be unknown)", I tried passing `LIQUID_MODEM_UNKNOWN` but `symtrack_cccf_create` fails here:
```c
if (_ms == LIQUID_MODEM_UNKNOWN || _ms >= LIQUID_MODEM_NUM_SCHEMES)
    return liquid_error_config("symtrack_%s_create(), invalid modulation scheme", EXTENSION_FULL);
```