jgaeddert/liquid-dsp #204: iirfilt_crcf_execute: how to change calculation method from df2 to df1

Cuda-Chen, 2020-06-09:
After browsing the source code, I found `iirfilt_crcf_execute` uses df2 method by default when filter format is sos. However, I would like to use df1 method in my case.

I came up with creating an `iirfiltsos` object instead of `iirfilt` but compiler complains about no such `iirfiltsos` object:
```
$ gcc iirfiltsos.c -lliquid -lm
iirfiltsos.c: In function ‘main’:
iirfiltsos.c:31:5: error: unknown type name ‘iirfiltsos_crcf’; did you mean ‘iirfilt_crcf’?
   31 |     iirfiltsos_crcf q = iirfiltsos_crcf_create(b, a);
      |     ^~~~~~~~~~~~~~~
      |     iirfilt_crcf
iirfiltsos.c:31:25: warning: implicit declaration of function ‘iirfiltsos_crcf_create’; did you mean ‘iirfilt_crcf_create’? [-Wimplicit-function-declaration]
   31 |     iirfiltsos_crcf q = iirfiltsos_crcf_create(b, a);
      |                         ^~~~~~~~~~~~~~~~~~~~~~
      |                         iirfilt_crcf_create
iirfiltsos.c:34:5: warning: implicit declaration of function ‘iirfiltsos_crcf_print’; did you mean ‘iirfilt_crcf_print’? [-Wimplicit-function-declaration]
   34 |     iirfiltsos_crcf_print(q);
      |     ^~~~~~~~~~~~~~~~~~~~~
      |     iirfilt_crcf_print
iirfiltsos.c:45:9: warning: implicit declaration of function ‘iirfiltsos_crcf_execute_df1’; did you mean ‘iirfilt_crcf_execute’? [-Wimplicit-function-declaration]
   45 |         iirfiltsos_crcf_execute_df1(q, x[i], &y[i]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         iirfilt_crcf_execute
iirfiltsos.c:48:5: warning: implicit declaration of function ‘iirfiltsos_crcf_destroy’; did you mean ‘iirfilt_crcf_destroy’? [-Wimplicit-function-declaration]
   48 |     iirfiltsos_crcf_destroy(q);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
      |     iirfilt_crcf_destroy
```
(The code snippet is here: https://github.com/Cuda-Chen/liquid-dsp-practice/blob/master/iirfiltsos/iirfiltsos.c)

So I want to know if there are some methods to use df1 method by either creating an `iirfiltsos` object correctly or choosing df1 method when calling `iirfilt_crcf_execute`?

fsheikh, 2020-06-09:
@Cuda-Chen I am afraid this is because the second order section API is missing from external header. Can you please see if this patch fixes your compilation problem.

Cuda-Chen, 2020-06-09:
@fsheikh The above patch fixes the problem and I successfully use df1 to execute.
Thanks for your help!

PR #205 (2021-05-20): iirfilt: Add missing SOS API for iirfilt object
Summary: IIRFILT second order section API is missing from external liquid header.
Fixes #204
commit 886e0547 (2021-05-20): iirfilt: Add missing SOS API for iirfilt object

Summary: IIRFILT second order section API is missing from external
header. Removed from internal header.
Fixes #204
closed 2021-05-20 by jgaeddert