jgaeddert/liquid-dsp #206: how can I get the phase error between a real signal and nco output?

hzsch, 2020-06-23:
The lib of pll part only has complex signal , but I have to use the real signal. Can I have some help?

hzsch, 2020-06-23:
I already try some methods,but can not get the correct result.
[image]

jgaeddert, 2020-06-27:
If the input signal is real-valued, you can just re-define what the phase error looks like. Here's a stand-alone example:

```C
// pll_real_example.c : simulation of a real-valued phase-locked loop

#include <stdio.h>
#include <stdlib.h>
#include <math.h>

int main() {
    // parameters and simulation options
    float        phase_in     =  3.00f;     // carrier phase offset
    float        frequency_in =  0.40;      // carrier frequency offset
    float        alpha        =  0.002f;    // PLL bandwidth
    unsigned int n            =   400;      // number of samples

    // initialize states
    float beta          = sqrtf(alpha);     // phase adjustment factor
    float phase_out     = 0.0f;             // output signal phase
    float frequency_out = 0.45f;            // output signal frequency

    // print line legend to standard output
    printf("# %6s %12s %12s %12s %12s %12s\n",
            "index", "input", "output", "phase error", "filtered", "freq. error");

    // run basic simulation
    unsigned int i;
    for (i=0; i<n; i++) {
        // compute input and output signals
        float signal_in  = sinf(phase_in);
        float signal_out = sinf(phase_out);

        // compute phase error estimate
        float error_signal = signal_in * cosf( phase_out );

        // apply loop filter and correct output phase and frequency
        frequency_out += alpha * error_signal;    // adjust phase
        phase_out     +=  beta * error_signal;    // adjust frequency

        // plotting: phase error (modulo pi)
        float phase_error = phase_in - phase_out;
        while (phase_error >  M_PI) phase_error -= M_PI;
        while (phase_error < -M_PI) phase_error += M_PI;

        // plotting: frequency error
        float freq_error = frequency_in - frequency_out;

        // print results to standard output for plotting
        printf("  %6u %12.8f %12.8f %12.8f %12.8f %12.8f\n",
                  i, signal_in, signal_out, error_signal, phase_error, freq_error);

        // increment input and output phase values
        phase_in  += frequency_in;
        phase_out += frequency_out;
    }
    return 0;
}
```

closed 2020-06-27 by jgaeddert

hzsch, 2020-06-27:
Thank you very much for your answer. It helped me a lot and solved the problem that bothered me for a long time.

 

 // pll_real_example.c : simulation of a real-valued phase-locked loop #include <stdio.h&gt; #include <stdlib.h&gt; #include <math.h&gt; int main() {     // parameters and simulation options     float        phase_in     =  3.00f;     // carrier phase offset     float        frequency_in =  0.40;      // carrier frequency offset     float        alpha        =  0.002f;    // PLL bandwidth     unsigned int n            =   400;      // number of samples     // initialize states     float beta          = sqrtf(alpha);     // phase adjustment factor     float phase_out     = 0.0f;             // output signal phase     float frequency_out = 0.45f;            // output signal frequency     // print line legend to standard output     printf("# %6s %12s %12s %12s %12s %12s\n",             "index", "input", "output", "phase error", "filtered", "freq. error");     // run basic simulation     unsigned int i;     for (i=0; i<n; i++) {         // compute input and output signals         float signal_in  = sinf(phase_in);         float signal_out = sinf(phase_out);         // compute phase error estimate         float error_signal = signal_in * cosf( phase_out );         // apply loop filter and correct output phase and frequency         frequency_out += alpha * error_signal;    // adjust phase         phase_out     +=  beta * error_signal;    // adjust frequency         // plotting: phase error (modulo pi)         float phase_error = phase_in - phase_out;         while (phase_error &gt;  M_PI) phase_error -= M_PI;         while (phase_error < -M_PI) phase_error += M_PI;         // plotting: frequency error         float freq_error = frequency_in - frequency_out;         // print results to standard output for plotting         printf("  %6u %12.8f %12.8f %12.8f %12.8f %12.8f\n",                   i, signal_in, signal_out, error_signal, phase_error, freq_error);         // increment input and output phase values         phase_in  += frequency_in;         phase_out += frequency_out;     }     return 0; }