jgaeddert/liquid-dsp #209: Where is "liquid_float_complex" defined?

Mabdalshakour, 2020-08-09:
Hi,

I am trying to use the library on windows in an FM transmitter MS Visual Studio project. I just copied the FM source files "freqmod.c" and "freqdem.c" and I am trying to include all required dependencies. I was stuck at the "liquid_xxxxx_complex" for double and floats, etc
According to the documentation at liquidsdr.org/doc/datastuctures/ this data type is just a placeholder for the float complex and double comlex. 
How can I define it? I can't see it defined anywhere in the source code

Thank you,
Regards,
M. Osman

GkvJeep, 2020-08-09:
MS not support c99 standard.

Mabdalshakour, 2020-08-10:
Thank you so much. Sorry I am kinda newbie. Then I should use MinGW for example, right?
I won't need to define liquid_float_complex by using it?

GkvJeep, 2020-08-10:
https://github.com/cjcliffe/CubicSDR/tree/master/external/liquid-dsp

Mabdalshakour, 2020-08-12:
Thank you for your help.
After using MinGW for gcc compiler and C99, I am still getting errors in the frequency modulation/demodulation examples. for example:
[image]

Do you know of any well tested procedure to compile the source code in windows?

Thank you,

vsonnier, 2020-08-12:
@Mabdalshakour The only procedure is the one I've written and can be be found at the place @GkvJeep indicated. The question is, do you really need to re-compile it ? Why not using the precompiled binaries ? They should work out of the box with any Visual Studio and Windows version. (the only dependency is MSVCRT.dll)

Mabdalshakour, 2020-08-13:
Thank you. I think I will just explain what I am doing so that you can guide me.
I have an SDR platform that I am writing a Visual studio C program for it. It has many APIs in the project. What I want is to just integrate the frequency modulator from the Liquid-DSP open source library in my project.
What I understand is the the liquid-DSP  is written in C99 for the gcc compiler. So what should I do in order to be able to compile it in the Visual Studio project?
If the liquid project is written in C, why I have many syntax errors even with MinGW?

Thank you very much

vsonnier, 2020-08-13:
Hi  @Mabdalshakour, 

> What I understand is the the liquid-DSP is written in C99 for the gcc compiler. So what should I do in order to be able to compile it in the Visual Studio project?

There is nothing you can do. You need MSYS2 to compile it, or use the CubicSDR pre-compiled binaries as I explained before. 

You'll only need the `libliquid.dll` at runtime. 

To develop in Visual Studio you'll need to use theCubicSDR  `include/liquid.h` and reference the `libliquid.lib` in your VS project. That's all.  You cannot compile the lib itself in Visual Studio, but can perfectly use the lib in a Visual Studio project. 

CubicSDR on Windows is compiled with VS2015/17/19 using this MSYS2-compiled liquid-dsp with no problem. The only drawback is that you can't step in liquid-dsp code from the Visual Studio debbugger. 
  
> If the liquid project is written in C, why I have many syntax errors even with MinGW?

Because this "portable C" uses POSIX functions only available in *Nix and MacOS (or MSYS2 environnement), and because the build system is based on `make`  which assumes a *Nix environment with GCC-named libraries and such.

There was an attempt to port the build system to CMake, but is was never finished because it never gained enough popularity.

vsonnier, 2020-08-14:
@Mabdalshakour  FYI I've updated the pre-compiled Windows libraries to the latest liquid-dsp `master` : https://github.com/cjcliffe/CubicSDR/commit/d08b0be9bdf82ce687a96cc25db4ea8304e21448

brian-armstrong, 2020-12-15:
I actually have a branch that builds in Windows/MSVC. It fixes this issue and others basically by making liquid look a bit more C++ compatible in places. It passes all tests but unfortunately it has drifted over time. https://github.com/quiet/quiet-dsp/tree/win

bignick26, 2022-07-01:
@brian-armstrong how did you handle the conflict with MCVC C++ Complex definition/standard? I am currently using @vsonnier's CubicSDR build, which has been worked awesome until I tried to use C++ and complex numbers.