jgaeddert/liquid-dsp #210: symtrack: unused member variables + decision-directed feeback for eqlms

pfeatherstone, 2020-08-13:
1) The following member variables are not used:
```
loat           agc_bandwidth;      // agc bandwidth
float           eq_bandwidth;       // equalizer bandwidth
float           pll_bandwidth;      // phase-locked loop bandwidth
```

2) You make a comment in symtrack.c to use decision-directed feedback in the equalizer. Is it as simple as replacing:

```
       MODEM(_demodulate)(_q->demod, d_hat, &sym_out);
        float phase_error = MODEM(_get_demodulator_phase_error)(_q->demod);

        // update equalizer independent of the signal: estimate error
        // assuming constant modulus signal
        // TODO: use decision-directed feedback when modulation scheme is known
        // TODO: check lock conditions of previous object to determine when to run equalizer
        if (_q->num_syms_rx > 200)
            EQLMS(_step)(_q->eq, d_hat/cabsf(d_hat), d_hat);
```

with 

```
        TO d_prime;
        MODEM(_demodulate)(_q->demod, d_hat, &sym_out);
	MODEM(_get_demodulator_sample)(_q->demod, &d_prime);
        float phase_error = MODEM(_get_demodulator_phase_error)(_q->demod);
	//float evm = MODEM(_get_demodulator_evm)(_q->demod);

        if (_q->num_syms_rx > 200)
            EQLMS(_step)(_q->eq, d_prime, d_hat);
```

pfeatherstone, 2020-09-16:
The fresh commits resolve this.

closed 2020-09-16 by pfeatherstone

dj0abr, 2020-11-11:
I'm using the latest code (11.2020).
The symbol tracker works fine when I do a symtrack.._reset() as soon as a signal is there.
When it runs on noise and after a while gets a signal again, it cannot sync by itself. I have to do a new reset to get it syncing.
Is this the normal behaviour, or is there anything do to the get it synchonized without an additional reset ?