jgaeddert/liquid-dsp #211: Wrong install path for lib: /usr/local/usr/local/lib

Zumili, 2020-09-08:
Hey guys, just tried to install this piece of software for a tool called **doppler** on Raspberry Pi 3 B+ (Raspbian GNU/Linux 10 (buster)), and got some problems during the installation.

Building works just fine,
```
git clone git://github.com/jgaeddert/liquid-dsp.git
cd liquid-dsp
./bootstrap.sh
./configure
make
```
but when it comes to
`sudo make install`
the path for the library is created wrong in the makefile.

Have a look at the output:
```
pi@raspberrypi:~/liquid-dsp $ sudo make install
installing...

mkdir -p /usr/local/include
mkdir -p /usr/local/include/liquid
mkdir -p /usr/local/usr/local/lib
#install -m 644 -p libliquid.so libliquid.a /usr/local/lib
install -m 644 -p libliquid.a libliquid.so /usr/local/usr/local/lib
install -m 644 -p include/liquid.h /usr/local/include/liquid

---------------------------------------------------------
  liquid-dsp was successfully installed.

  On some machines (e.g. Linux) you should rebind your
  libraries by running 'ldconfig' to make the shared
  object available.  You might also need to modify your
```

The path which is created looks like this:  /usr/local/usr/local/lib
but should look like this:  /usr/local/lib

It looks that the problem occurs
in https://github.com/jgaeddert/liquid-dsp/blob/master/makefile.in#L1232
and https://github.com/jgaeddert/liquid-dsp/blob/master/makefile.in#L1234
and also in the uninstall part https://github.com/jgaeddert/liquid-dsp/blob/master/makefile.in#L1257

The problem seems to be with this term:
 `$(DESTDIR)$(exec_prefix)$(libdir)`

Should it not look like this?:
 `$(DESTDIR)$(exec_prefix)/lib`

Can somebody confirm this problem?

Greetings Thomas

jgaeddert, 2020-09-08:
Yes! Thanks for pointing this out. I just discovered this today. It’s the result of a botched merge. Looking into it now!

jgaeddert, 2020-09-09:
Should be fixed in c216570afba4c6ef962e90529c63fae6ef20b483

closed 2020-09-09 by jgaeddert