jgaeddert/liquid-dsp #214: 8-PSK Tracking

dj0abr, 2020-11-02:
First I want to thank you for this awsome library, it's maybe the best SDR lib available today.
I have implemented a QPSK and 8PSK transmission via soundcard.
On the receiver side I use symtrack for sync and symbol recovery. This works perfectly in QPSK.
When it comes to 8PSK then it is syncing, I can see the constellation points in the diagram, but these points are slowly rotating clockwise.
Since I have simply connected line-In to line-out of my soundcard, the clock is the same for RX and TX. So I think this rotation is cause by symtrack. 
What do you think is the best way to find a solution ?
Kurt

jgaeddert, 2020-11-10:
Hey, thanks for the nice comments, Kurt! I wonder if the symtrack isn't being configured properly to receive 8-PSK. How did you initialize it? You can also use the `set_modscheme()` method and explicitly specify the modulation scheme to use. For example:

```C
symtrack_crcf sync = symtrack_crcf_create_default(); // 2 samples/symbol and QPSk by default
symtrack_crcf_set_modscheme(sync, LIQUID_MODEM_PSK8);
```

dj0abr, 2020-11-11:
thanks jgaeddert ! I am now using 8APSK which works fine, no rotation. Will go back to 8PSK later, in the meantime I have an issue with symtrack, but will ask in a new thread.

closed 2020-12-18 by dj0abr