jgaeddert/liquid-dsp #216: Question : how do you do carrier frequency recovery with real IQ?

pfeatherstone, 2020-11-25:
Doing carrier frequency recovery with complex IQ is straight forward.
Either do:
```
zn -> zn * exp(i(dphi*n + phi))
```
or use ```nco_crcf```
How do you do it if you have real IQ without doing a Hilbert transform first?

jgaeddert, 2020-12-02:
Are you asking about a purely real phase-locked loop? The principle is the same as with a complex one, but you're mixing with a real-valued signal; this creates an image that you need to filter off.

If you're asking how to down-convert a real signal, the principle is also the same as with a complex one, but you'd need to filter off the image using a low-pass filter after mixing.

pfeatherstone, 2020-12-02:
Thank you for the clarification.

closed 2020-12-02 by pfeatherstone