jgaeddert/liquid-dsp #219: How can I transmit/receive complex signal?

wondoh, 2021-01-08:
Hi,

I know this would be a stupid question.

I would like to transmit modulated data over 16bit PCM audio signal (Fs=48kHz), and demodulate it from the audio signal.
My workflow is:
data -> modulation -> [audio] -> audio encoding -> IP network -> audio decoding -> [audio] -> demodulation -> data

        float complex x
        modem_modulate(mod, i, &x); 
        **// how can i convert complex x to uint16?
        // and convert from uint16 to complex x?**
        modem_demodulate(demod, x, &s);

I browsed all the examle codes but couldn't find any clue using your library.

Thank you.

jgaeddert, 2021-01-09:
Great question! There are really two steps to this:

  1. convert from complex baseband to an intermediate frequency (IF), real-valued samples
  2. convert from floating point to fixed point

The first step can be achieved with the [Hilbert Transform](https://liquidsdr.org/doc/firhilb/). So for example, if you have a signal at complex baseband (f=0) sampled at 24 kHz complex, and run it through `firhilbf_interp_execute` you will get a real-valued output sampled at 48 kHz real and the signal centered at 12 kHz. Other frequencies are certainly possible, of course, but this is the most computationally efficient.

Once you have `float` values, convert to `uint16` however is appropriate for your sound card. This might require scaling and adding an offset before performing a type conversion (e.g. `uint16_t u = (uint16_t) (sample*1000.0f + (1<<15))`).

Demodulation is just the reverse of this.

dj0abr, 2021-01-18:
I just did a project sending/receiving modulated audio data. I used the liquid NCO's up/down mixer to bring it to 12 kHz (in my case 1,5kHz). Followed by the formular  output = sqrt(re² + im²) to get the samples for the soundcard.
Your answer to use a Hilbert Transform sounds interesting. Is there any advantage/disadvantage compared with the NCO-Mixer solutuion, or is the result almost the same. Maybe there is a difference in cpu usage ?

jgaeddert, 2021-01-23:
The Hilbert transform is a convenient way to convert from real to complex with the benefit of mixing up by Fs/4 essentially for free (moving Fs/4 around the unit circle is equivalent to multiplying by +1, +j, -1, -j in which case you don't really need to perform any actual multiplications!). When you need other center frequencies or sample rates, there are efficient ways to do that as well, but it's not as clear cut and takes a little bit more planning.