jgaeddert/liquid-dsp #220: Question : channel : adding multipath : why use msequence ?

pfeatherstone, 2021-01-14:
Looking at the source code for ```channel```, when adding multi-path, why use:
```
msequence_create_default(14);
```
?

Can you not use any uniformally distributed complex number in [-1-1i,1+1i] ?

Using ```msequence``` always gives you the same channel profile for a given multi-path filter length.

pfeatherstone, 2021-01-14:
The function for creating a "multi-path filter" is roughly:
```
firfilt_cccf create_multi_path_filter(int mptap)
{
    std::vector<float> mph(mptap);
    mph[0] = 1.0f;
    msequence ms = msequence_create_default(14);
    for (size_t i = 1; i < mph.size(); i++) {
        float vi = msequence_generate_symbol(ms, 8) / 256.0f - 0.5f;
        float vq = msequence_generate_symbol(ms, 8) / 256.0f - 0.5f;
        mph[i] = complex<float>(vi, vq) * 0.5f;
    }
    msequence_destroy(ms);
    return firfilt_cccf_create(&mph[0], mph.size());
}
```
which always give you the same filter for a given number of taps.

pfeatherstone, 2021-01-15:
@brian-armstrong I've seen from issue activity that you are an avid user. Can you help with this question?

brian-armstrong, 2021-01-15:
I can't say I have any idea. I'm a software engineer that just pretends to know DSP sometimes. I've seen some others around here who do seem to know and of course Joseph will.

pfeatherstone, 2021-01-15:
Okidok. Thanks anyway. I'll wait for Joseph's response.

fsheikh, 2021-01-17:
My two cents, I/Q components of channel should be normally (Gaussian) distributed to mimic Rayleigh fading [channels](http://www.wirelesscommunication.nl/reference/chaptr03/rayleigh.htm). Now `m_sequence` is pseudo-random of course but [documentation](https://liquidsdr.org/doc/msequence/) suggests the generated sequence has very good auto-correlation properties, possibly to reduce inter-symbol interference effects. Having said that, original considerations might be different, in which case I will more than happy to be corrected :innocent:

jgaeddert, 2021-01-17:
@fsheikh is right; Gauss random variables are generally more useful for generating fading channels. I can't remember off the top of my head why I used uniform. Certainly using the msequence generator makes generating a random channel repeatable. Sounds like it's time for me to dust off the code I started to generate thread-safe, pseudo random number generators.

pfeatherstone, 2021-01-17:
So will any Gaussian distributed random numbers do then ?

pfeatherstone, 2021-01-20:
`I can't remember off the top of my head why I used uniform`
@jgaeddert Was that because you get enforced clipping in range [-0.25,0.25]. Whereas if  you use gaussian distributions, that's not true and you can get very large filter taps however unlikely.
If that's the case, can you just clip appropriately and use the following in C++ for example:
```
std::default_random_engine generator;
std::normal_distribution<float> distribution(0.0,0.25);
for (...)
    complex<float> tap = complex<float>(distribution(generator),distribution(generator));
```

pfeatherstone, 2021-01-20:
It would be great to be able to simulate different multi-path channel profiles given the same number of filter taps.

jgaeddert, 2021-01-23:
> So will any Gaussian distributed random numbers do then ?

It really depends on the channel you are trying to model. There have been a ton of measurement campaigns to characterize wireless channels, and it depends on many variables (carrier frequency, mobility, indoor/outdoor, urban/rural, short-range/long-range, materials in the environment, antenna pattern, atmospheric variability, and of course the geometry of the transmitter/receiver).

In general, channel modeling can be broadly characterized into bins (flat fading vs. frequency-selective fading, static vs. time varying, etc.) and the specifics of each depend on the environment. At a high level, you can think of filter coefficients as reflections off objects with a delay, phase, and attenuation. In reality, though, it's much more complicated due to the materials involved. For example, a wide-band RF pulse bounced off a piece of concrete is heavily distorted due to the composition of the concrete itself; it's not just a simple delay and attenuation.

> @jgaeddert Was that because you get enforced clipping in range [-0.25,0.25]. Whereas if you use gaussian distributions, that's not true and you can get very large filter taps however unlikely.

I doubt it was because of clipping. Most likely it was laziness on my part! 😬

pfeatherstone, 2021-01-24:
@jgaeddert Wow, thanks for the detailed answer! Do you have any sources on how to characterize multi-path numerically using different models? I would love to enrich some siggen code for testing algorithms. Thanks