jgaeddert/liquid-dsp #221: No matching function

Hepan02, 2021-01-15:
Hi,
I have already install liquid with successes. Examples work properly. 
But if I create new cpp script with liquid function I am not able to compile it with cmake and even gcc.

I have errors like:
```
root@beaglebone:/home/debian/workarea/uhd-3.13.1.0/host/examples/init_usrp/build# **make**
[ 50%] Building CXX object CMakeFiles/init_usrp.dir/init_usrp.cpp.o
/home/debian/workarea/uhd-3.13.1.0/host/examples/init_usrp/init_usrp.cpp:42:16: error: **no matching function** for call to 'modem_create'
        modem mod   = modem_create(ms);
                      ^~~~~~~~~~~~
/usr/local/include/liquid/liquid.h:2492:7: note: candidate function not viable: requires 2 arguments, but 1 was provided
modem modem_create(modulation_scheme, unsigned int _bits_per_symbol);
      ^
/home/debian/workarea/uhd-3.13.1.0/host/examples/init_usrp/init_usrp.cpp:43:19: error: no matching function for call to 'modem_create'
    modem demod = modem_create(ms);
                  ^~~~~~~~~~~~
/usr/local/include/liquid/liquid.h:2492:7: note: candidate function not viable: requires 2 arguments, but 1 was provided
modem modem_create(modulation_scheme, unsigned int _bits_per_symbol);
      ^
2 errors generated.
make[2]: *** [CMakeFiles/init_usrp.dir/build.make:63: CMakeFiles/init_usrp.dir/init_usrp.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/init_usrp.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```
I am not able also to compile simple code with gcc or clang. Even if I try to compile the example I have errors.
I use cmake because I will create script with liquid and UHD fof my usrp. 

Should I link library? What can I do? May you help me?

# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 2 (v7l)
BogoMIPS        : 995.32
Features        : half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc08
CPU revision    : 2

Hardware        : Generic AM33XX (Flattened Device Tree)
Revision        : 0000
Serial          : 4219BBBK05E9

Hepan02, 2021-01-15:
My CMakeLists.txt

`
#
# Copyright 2014-2015 Ettus Research LLC
# Copyright 2018 Ettus Research, a National Instruments Company
#
# SPDX-License-Identifier: GPL-3.0
#

cmake_minimum_required(VERSION 2.8)
include_directories(include /usr/local/include/liquid/)

### Set up build environment ##################################################
# Choose a static or shared-library build (shared is default, and static will
# probably need some special care!)
# Set this to ON in order to link a static build of UHD:
option(UHD_USE_STATIC_LIBS OFF)

# This example also requires Boost:
set(BOOST_REQUIRED_COMPONENTS
    program_options
    system
    thread
)
if(MSVC)
    set(BOOST_ALL_DYN_LINK "${BOOST_ALL_DYN_LINK}" CACHE BOOL "boost enable dynamic linking")
    if(BOOST_ALL_DYN_LINK)
        add_definitions(-DBOOST_ALL_DYN_LINK) #setup boost auto-linking in msvc
    else(BOOST_ALL_DYN_LINK)
        set(BOOST_REQUIRED_COMPONENTS) #empty components list for static link
    endif(BOOST_ALL_DYN_LINK)
endif(MSVC)
find_package(Boost "1.46" REQUIRED ${BOOST_REQUIRED_COMPONENTS})

# To add UHD as a dependency to this project, add a line such as this:
find_package(UHD "3.8.0" REQUIRED)
# The version in  ^^^^^  here is a minimum version.
# To specify an exact version:
#find_package(UHD 3.8.1 EXACT REQUIRED)

### Configure Compiler ########################################################
if(CMAKE_VERSION VERSION_LESS "3.1")
    if(${CMAKE_CXX_COMPILER_ID} STREQUAL "GNU")
        set(CMAKE_CXX_FLAGS "--std=gnu++11 ${CMAKE_CXX_FLAGS}")
    elseif(${CMAKE_CXX_COMPILER_ID} STREQUAL "Clang")
        if("${IS_APPLE}" STREQUAL "")
            set(CMAKE_CXX_FLAGS "-std=c++11 ${CMAKE_CXX_FLAGS}")
        else()
            set(CMAKE_CXX_FLAGS "-std=c++11 -stdlib=libc++ ${CMAKE_CXX_FLAGS}")
        endif()
    endif()
else()
    set(CMAKE_CXX_STANDARD 11)
endif()

if(CMAKE_SYSTEM_NAME STREQUAL "FreeBSD" AND ${CMAKE_CXX_COMPILER_ID} STREQUAL "Clang")
    set(CMAKE_EXE_LINKER_FLAGS "-lthr ${CMAKE_EXE_LINKER_FLAGS}")
    set(CMAKE_CXX_FLAGS "-stdlib=libc++ ${CMAKE_CXX_FLAGS}")
endif()

include_directories(
    ${Boost_INCLUDE_DIRS}
    ${UHD_INCLUDE_DIRS}
)
link_directories(${Boost_LIBRARY_DIRS})

### Make the executable #######################################################
add_executable(init_usrp init_usrp.cpp)

set(CMAKE_BUILD_TYPE "Release")
message(STATUS "******************************************************************************")
message(STATUS "* NOTE: When building your own app, you probably need all kinds of different  ")
message(STATUS "* compiler flags. This is just an example, so it's unlikely these settings    ")
message(STATUS "* exactly match what you require. Make sure to double-check compiler and     ")
message(STATUS "* linker flags to make sure your specific requirements are included.          ")
message(STATUS "******************************************************************************")

# Shared library case: All we need to do is link against the library, and
# anything else we need (in this case, some Boost libraries):
if(NOT UHD_USE_STATIC_LIBS)
    message(STATUS "Linking against shared UHD library.")
    target_link_libraries(init_usrp ${UHD_LIBRARIES} ${Boost_LIBRARIES})
# Shared library case: All we need to do is link against the library, and
# anything else we need (in this case, some Boost libraries):
else(NOT UHD_USE_STATIC_LIBS)
    message(STATUS "Linking against static UHD library.")
    target_link_libraries(init_usrp
        # We could use ${UHD_LIBRARIES}, but linking requires some extra flags,
        # so we use this convenience variable provided to us
        ${UHD_STATIC_LIB_LINK_FLAG}
        # Also, when linking statically, we need to pull in all the deps for
        # UHD as well, because the dependencies don't get resolved automatically
        ${UHD_STATIC_LIB_DEPS}
    )
endif(NOT UHD_USE_STATIC_LIBS)

### Once it's built... ########################################################
# Here, you would have commands to install your program.
# We will skip these in this example.
`

JayKickliter, 2021-01-15:
The error says everything you need to know. You are calling `modem_create` like
```c
modem mod   = modem_create(ms);
```
but it requires two parameters, modulation scheme (which I assume is what your `ms` argument), and bits/symbol which you are not providing:

```c                                 
modem mod  modem_create(ms, /* add an argument for bits per symbol here */ );
```

Hepan02, 2021-01-17:
@JayKickliter  thank you. It helped me. But I still have problem with compiling.

`
root@beaglebone:/home/debian/workarea/uhd-3.13.1.0/host/examples/init_usrp/build# make
Scanning dependencies of target init_usrp
[ 50%] Building CXX object CMakeFiles/init_usrp.dir/init_usrp.cpp.o
[100%] Linking CXX executable init_usrp
/usr/bin/ld: CMakeFiles/init_usrp.dir/init_usrp.cpp.o: in function `main':
init_usrp.cpp:(.text+0x90): undefined reference to `windowf_create'
/usr/bin/ld: init_usrp.cpp:(.text+0xa0): undefined reference to `modem_create'
/usr/bin/ld: init_usrp.cpp:(.text+0xb0): undefined reference to `modem_create'
/usr/bin/ld: init_usrp.cpp:(.text+0xf0): undefined reference to `windowf_push'
/usr/bin/ld: init_usrp.cpp:(.text+0x124): undefined reference to `windowf_push'
/usr/bin/ld: init_usrp.cpp:(.text+0x158): undefined reference to `windowf_push'
/usr/bin/ld: init_usrp.cpp:(.text+0x18c): undefined reference to `windowf_push'
/usr/bin/ld: init_usrp.cpp:(.text+0x1c0): undefined reference to `windowf_push'
/usr/bin/ld: CMakeFiles/init_usrp.dir/init_usrp.cpp.o:init_usrp.cpp:(.text+0x1f4): more undefined references to `windowf_push' follow
/usr/bin/ld: CMakeFiles/init_usrp.dir/init_usrp.cpp.o: in function `main':
init_usrp.cpp:(.text+0x2d0): undefined reference to `windowf_read'
/usr/bin/ld: init_usrp.cpp:(.text+0x4f8): undefined reference to `windowf_destroy'
/usr/bin/ld: init_usrp.cpp:(.text+0x500): undefined reference to `modem_destroy'
/usr/bin/ld: init_usrp.cpp:(.text+0x508): undefined reference to `modem_destroy'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/init_usrp.dir/build.make:91: init_usrp] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/init_usrp.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`

`
My library:
#include <uhd/utils/thread.hpp>
#include <uhd/utils/safe_main.hpp>
#include <uhd/utils/static.hpp>
#include <uhd/usrp/multi_usrp.hpp>
#include <uhd/exception.hpp>
#include <boost/program_options.hpp>
#include <boost/math/special_functions/round.hpp>
#include <boost/format.hpp>
#include <boost/algorithm/string.hpp>
#include <stdint.h>
#include <iostream>
#include <csignal>
#include <string>
#include <chrono>
#include <thread>
#include <stdlib.h>
#include <stdio.h>
#include <getopt.h>
#include <liquid/liquid.h>
#include <complex>
`

What about _reference_ problem?

jgaeddert, 2021-01-17:
That's definitely a linker error. It seems as though it isn't finding libliquid.so

JayKickliter, 2021-01-17:
Try adding `target_link_libraries(<YOUR_TARGET_NAME>, liquid)`

Hepan02, 2021-01-18:
I added this line to CMakeLists.txt:

`
target_link_libraries(init_usrp liquid)
`

Where init_usrp is the name of my targed.
It helped, Now I am able to compile scripts.

Thank you for your help.

JayKickliter, 2021-03-29:
This can be closed.

closed 2021-05-21 by jgaeddert