jgaeddert/liquid-dsp #222: msourcecf_write_samples is working incorrectly

quanpm4900, 2021-01-25:
#define MAX_SAMPLES_NUM 4096
mSampleNum = 2048
mBuffer = (liquid_float_complex*)malloc(MAX_SAMPLES_NUM * sizeof(liquid_float_complex));
mSiggen = msourcecf_create_default();
msourcecf_add_tone(mSiggen, -0.4f, 0.09f, 20);
int k = msourcecf_write_samples(mSiggen, mBuffer, mSampleNum);
qDebug() << k;
when I run test, k is not equal mSampleNum (2048), k random below 1000

fsheikh, 2021-01-25:
Please see[here](https://github.com/jgaeddert/liquid-dsp/blob/master/src/framing/src/msource.c#L395), *write_samples* function returns an error/success code (return value should be zero in normal case) and NOT the configured buffer length

quanpm4900, 2021-01-25:
thank @fsheikh, return value not be zero so there is error, could you suggest me to print this error?

quanpm4900, 2021-01-25:
@jgaeddert could you add makefile.mingw32 and makefile.mingw64 for building dll on windows?

jgaeddert, 2021-01-30:
Building on Windows is a pretty common request; although I don't have easy access to a Windows system for testing. And I don't have a lot of experience developing in Windows in general.