jgaeddert/liquid-dsp #224: ./configure --prefix=     not possible?

jrmims, 2021-02-16:
I'm trying to build liquid dsp in specific directories using` --prefix=<path to directory>`. I'm getting the error ```make: *** No rule to make target `src/libliquid.o', needed by `libliquid.a'.  Stop.``` when I run `make` in that directory.

jgaeddert, 2021-02-20:
What OS is this on?

jgaeddert, 2021-02-20:
I wasn't able to reproduce this on a Debian build. The `--prefix` option with `configure` should only affect the install directory. Can you try doing a full build from scratch? That is `make distclean` and then re-run `./bootstrap.sh && ./configure --prefix=/path/to/install/directory && make`?

jrmims, 2021-02-26:
Thanks for getting back to me! I believe the issue was I was running configure from inside the install directory with `../configure --prefix=/path && make`

Running everything from inside the base liquiddsp directory worked perfectly!

closed 2021-02-26 by jrmims