jgaeddert/liquid-dsp #225: Question regarding flexframegen

mryndzionek, 2021-03-13:
I'm trying to use the `flexframegen` object with different modulation schemes. When I look at a IQ waterfall the signals look very similar regardless of the modulation scheme used:
BPSK
[image]

QPSK
[image]

ASK2
[image]

Is this supposed to be this way? BPSK and ASK2 look almost the same...

pfeatherstone, 2021-03-13:
BPSK and ASK2 are the same

pfeatherstone, 2021-03-13:
Somewhere in the liquid docs there will be some pictures of the different constellations. There you will see BPSK is the same as ASK2, QPSK is the same as PSK4 (maybe there is a phase difference of pi/4,  it sure)

mryndzionek, 2021-03-13:
Okay, I've found it: https://liquidsdr.org/doc/modem/

closed 2021-03-13 by mryndzionek