jgaeddert/liquid-dsp #230: spwaterfallcf_example: spwaterfallcf_set_commands() fails

pfeatherstone, 2021-05-19:
In ```spwaterfallcf_example```, ```spwaterfallcf_set_commands()``` fails with:
```
error [3]: invalid parameter or configuration
  src/fft/src/spwaterfall.c:263: spwaterfallcf_set_commands(), input string size exceeds reasonable limits
```

pfeatherstone, 2021-05-19:
Ah, easy fix. Line 261 in ```spwaterfall.c```, the if statement isn't surrounded by curly brackets. So the error is always "thrown". I won't submit a PR, the fix is too trivial. @jgaeddert

JayKickliter, 2021-05-19:
> I won't submit a PR, the fix is too trivial

That’s the perfect kind of fix for a PR. Low work for you, low review burden.

pfeatherstone, 2021-05-19:
There are 27 open PRs, i don't think they are getting merged very often...

pfeatherstone, 2021-05-19:
Okidok

fsheikh, 2021-05-19:
>  Many of the PRs on this project, however, are changes that don't necessarily fit with the original intent of the library

well, at least I would love to hear how [these](https://github.com/jgaeddert/liquid-dsp/pulls/fsheikh) collide with "*original intent*"

jgaeddert, 2021-05-20:
@fsheikh fair point

jgaeddert, 2021-05-23:
Seems like there are several issues if I enable strict exit on error, but for now this particular issue should be fixed with 4f3572ce02173f9732a7db2cd9c349821cd0541b.

closed 2021-05-23 by jgaeddert

pfeatherstone, 2021-05-24:
It seems like you've approximated C++ exceptions with ```liquid_error```

jgaeddert, 2021-05-25:
That's right except C has no real way to handle exceptions except for manually checking the output of each function call (e.g. no clean way to try/except). `liquid_error` was an attempt to meet C++ halfway and provide some error checking while staying thread safe. It still has a ways to go in that regard though.

pfeatherstone, 2021-05-25:
It's funny, you hear loads of people moaning about how c++ exceptions are bad. Then you think really hard about how you could do error handling correctly, consistently which propagates upwards nicely, and you end up re-inventing exceptions for "exceptional" error handling...