jgaeddert/liquid-dsp #231: Question : FSK - what are "good" parameters.

pfeatherstone, 2021-05-24:
I've been playing with ```fskmodem_example.c``` in the examples folder.
I'm trying to understand how ```fskmod``` works.
So we have:
* ```m``` : bits per symbol
* ```k```: number of samples per symbol, typically equal to ```2^m``` but a good value is ```2^(m+1)```
* ```bw```: the cutoff frequency

So there are ```M = 2^m``` possible symbols in range ```0 <= s <= 2^m - 1```.
Each symbol occupies a slice of the bandwidth. So symbol ```s``` has normalized frequency ```f_s = (s - M2) * bw / M2```
where:
* ```M2 = (M-1)/2```
So symbol ```s=0``` has normalized frequency ```f = -bw``` and symbol ```s=M-1``` has normalized frequency ```f = +bw```.

Now it looks like each symbol ```s``` gets modulated by generating a tone of ```k``` samples at its normalized frequency, i.e
```f = (s - M2) * bw / M2```

What I don't understand is how ```k``` and ```bw``` affect the shape of the signal.

So if I use:
* ```m = 3;```
* ```k = 2 << m;``` (=16) (recommended value from the example)
* ```bw = 0.2``` (recommended value form the example)

I get the following signal:
[image]

If I use:
* ```m = 3;```
* ```k = 32;```
* ```bw = 0.2```

I get the following:
[image]

Now if I play with bandwidth:

* ```m=3;```
* ```k=16;``` (recommended)
* ```bw=0.1;```

I get:
[image]

* ```m=3;```
* ```k=32;```
* ```bw=0.1;```

I get:
[image]

* ```m=3;```
* ```k=64;```
* ```bw=0.1;```

I get:

[image]

So it looks like ```k``` is affecting the rolloff. Is that right?

What is causing the rippling?

How do I judge what a good value for ```k``` is given ```m``` and ```bw```?

@jgaeddert Thank you

pfeatherstone, 2021-05-24:
Ah, is the rippling on top basically the ```2 ^ 3 = 8``` (for ```m=3```) tones starting to appear?

pfeatherstone, 2021-05-24:
Is there a way to guarantee a flat top?

pfeatherstone, 2021-05-25:
@jgaeddert @brian-armstrong any ideas?

jgaeddert, 2021-05-25:
It's a tradeoff between error rate and bandwidth. Minimum spacing (which I think gives a flat top) is where the carrier spacing is equal to the symbol rate.

pfeatherstone, 2021-05-25:
So carrier spacing is ```bw / M = bw / (2^m)``` and symbol rate is ```bw/k```.
So we want ```k = 2^m``` ?

pfeatherstone, 2021-05-25:
So bandwidth is ```2*bw``` using your variable names (in the code you use ```bw``` for a cutoff frequency i think, hence the factor of 2)
So carrier spacing is ```2*bw / (2^m - 1)``` and symbol rate is ```1/k```.
So we want ```k = (2^m - 1) / (2*bw)```
So if ```bw = 0.25``` and bandwidth is 0.5 then ```k = 2 * (2 ^m - 1) = 2^(m+1) - 2```.
so we are close to the default value used in the example which is ```k = 2 << m = 2^(m+1)```

Is this right?

pfeatherstone, 2021-05-25:
Hmm doesn't give me a flat top.
I used ```m=3```, ```fc = 0.2```, then ``` k = (2^m - 1) / (2*fc) = 17.5 ~ 18```
Which gives me:
[image]

jgaeddert, 2021-05-26:
I need to double check the values. Those spikes are probably a result of not maintaining phase continuity between symbols.

jgaeddert, 2021-05-26:
More precisely, each like symbol is within phase.

pfeatherstone, 2021-05-29:
> I need to double check the values. Those spikes are probably a result of not maintaining phase continuity between symbols.

Does liquid not maintain phase continuity in fskmodem ?

jgaeddert, 2021-05-30:
So liquid absolutely maintains phase continuity between adjacent symbols. I think I mis-"spoke" with my previous comment. If all like symbols have the same phase then you'll see spikes in the spectral estimate. It all depends upon the parameters of the FSK modulator you've supplied.

closed 2021-06-27 by pfeatherstone