jgaeddert/liquid-dsp #233: Question: Simulating SDR impulse response

pfeatherstone, 2021-06-03:
Some SDRs have an impulse response such that the noise floor droops at the edges.
How can we simulate this?
So far i've passed the noise floor (generated using AWGN) through a firfilt object using a cutoff frequency of 0.9 and a filter of size 13, so has pretty slow rolloff. 
Is this a sensible thing to do? Is this physically meaningful? Should i use an iirfilt instead? In which case, what are good parameters for the iirfilt (it seems to have way more configuration than firfilt). Is a windowed sinc filter even a good thing to use? Can we more accurately determine the impulse response of a typical SDR without doing tons of measurements?
Thanks!!!

jgaeddert, 2021-06-04:
Great question! I think it depends on what the SDR itself is doing. The shape of the roll-off might look similar between a FIR and IIR filter; however the group delay will absolutely be different. Lumped components (capacitors, inductors, etc.) have digital IIR equivalents, but many SDRs use a digital filter basically embedded in the ADC itself. If you can generate a known wide-band input into the SDR's receiver, it's not too difficult to determine its linear response. But, as you point out, this can be a pain to measure.

I think your best bed would be to use a FIR filter with a windowed sinc (bandwidth of 0.9 is probably sufficient).

pfeatherstone, 2021-06-04:
Cheers. I will do that then.

closed 2021-06-27 by pfeatherstone