jgaeddert/liquid-dsp #238: Sample amplitude greater than 1.0

glv2, 2021-06-25:
I generate frame samples using ```gmskframegen```, their maximum amplitude is 1.0.
When resampling them with ```msresamp_crcf```, I noticed that some of the samples have an amplitude around 1.06 or 1.07.

It caused some issues with another library when converting the samples from cf32 format to cs8 format to send them to a SDR as it was expecting the maximum amplitude to be 1.0. I solved this by multiplying all the samples by 0.9.

Is it expected that this can happen, or is it a bug?

jgaeddert, 2021-06-25:
It's totally expected but I see why you would ask the question. GMSK is of course a nonlinear modulation type, but when you apply a linear band-limiting filter to it, you will add some amplitude modulation to the signal. This is the nature of linear filters. The exceeds amplitude will increase if you reduce the filter bandwidth (that is, cut off more of the GMSK signal). I don't have a clear way to predict what that overshoot should be, but if you scale by 0.9 you should probably be ok. You can always clip at 0.999 if you're worried about conversion throwing an exception. This will cause some amount of spectral regrowth but it shouldn't be drastic.

glv2, 2021-06-25:
Ok, thanks for the clarification.

closed 2021-06-25 by glv2