jgaeddert/liquid-dsp #240: Question : different between polyphase channelizer and STFT

pfeatherstone, 2021-06-27:
THIS ISN'T A BUG REPORT

What is the difference between a polyphase FFT channelizer and a short-time-fourier-transform (STFT)
Looking at the formulae, they seem to be doing the same thing...

STFT formula is:

X[ω,m]= k=0 ∑ win_length-1       window[k] * input[m×hop_length+k] exp(−i * 2π⋅ωk / win_length)

pfeatherstone, 2021-06-27:
In fact, if one sets ```hop_length = win_length```, then that is exactly a polyphase FFT channelizer right ?

pfeatherstone, 2021-06-29:
@jgaeddert have you had a chance to think about this?

jgaeddert, 2021-07-01:
Sorry for the delay; work has been especially busy as of late.

At first glance they might look similar; however the difference lies in the stride over which the inner product between the taper (or equivalently window in STFT) and the input is taken relative to the DFT. In the STFT, one first takes the inner product between the window and the input (with an offset) before taking its DFT. With a FFT polyphase channelizer one takes `M` *outer* products between the taper and the input (with an offset) every `M` samples, and then the DFT is taken on the resulting `M` outputs.

Side note: the difference between a taper and window is that a window cannot generally be negative while a taper can.