jgaeddert/liquid-dsp #246: Stack overflow in FIR Farrow filter

fsheikh, 2021-07-23:
clang address sanitizer caught a stack overflow FIR Farrow filter implementation. [This](https://github.com/jgaeddert/liquid-dsp/blob/master/src/filter/src/firfarrow.c#L315) array seems to be allocated one element less. It has to be `Q+1` otherwise matrix multiplication [later](https://github.com/jgaeddert/liquid-dsp/blob/master/src/filter/src/firfarrow.c#L335) would not work.

Please see detailed report below. This can result in random aborts while using FIR farrow filter.
```cpp
=================================================================
==17925==ERROR: AddressSanitizer: dynamic-stack-buffer-overflow on address 0x7ffee332bb14 at pc 0x00000057bb40 bp 0x7ffee332b400 sp 0x7ffee332b3f8
WRITE of size 4 at 0x7ffee332bb14 thread T0
#0 0x57bb3f in matrixf_mul /home/fsheikh/liquid-dsp/src/matrix/src/matrix.math.c:122:43
#1 0x55d982 in polyf_fit /home/fsheikh/liquid-dsp/src/math/src/poly.common.c:88:5
#2 0x4fc97f in firfarrow_rrrf_genpoly /home/fsheikh/liquid-dsp/src/filter/src/firfarrow.c:335:9
#3 0x4fbf7d in firfarrow_rrrf_create /home/fsheikh/liquid-dsp/src/filter/src/firfarrow.c:109:5
#4 0x4f65ca in main /home/fsheikh/liquid-dsp/examples/firfarrow_rrrf_example.c:50:24
#5 0x7fa97f972b96 in __libc_start_main /build/glibc-2ORdQG/glibc-2.27/csu/../csu/libc-start.c:310
#6 0x41b479 in _start (/home/fsheikh/liquid-dsp/examples/firfarrow_rrrf_example+0x41b479)

Address 0x7ffee332bb14 is located in stack of thread T0
SUMMARY: AddressSanitizer: dynamic-stack-buffer-overflow /home/fsheikh/liquid-dsp/src/matrix/src/matrix.math.c:122:43 in matrixf_mul
Shadow bytes around the buggy address:
0x10005c65d710: cb cb cb cb 00 00 00 00 00 00 00 00 00 00 00 00
0x10005c65d720: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x10005c65d730: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x10005c65d740: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x10005c65d750: 00 00 00 00 00 00 00 00 00 00 00 00 ca ca ca ca
=>0x10005c65d760: 00 00[04]cb cb cb cb cb ca ca ca ca 00 00 00 cb
0x10005c65d770: cb cb cb cb ca ca ca ca 00 00 00 cb cb cb cb cb
0x10005c65d780: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x10005c65d790: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x10005c65d7a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0x10005c65d7b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
Addressable: 00
Partially addressable: 01 02 03 04 05 06 07
Heap left redzone: fa
Freed heap region: fd
Stack left redzone: f1
Stack mid redzone: f2
Stack right redzone: f3
Stack after return: f5
Stack use after scope: f8
Global redzone: f9
Global init order: f6
Poisoned by user: f7
Container overflow: fc
Array cookie: ac
Intra object redzone: bb
ASan internal: fe
Left alloca redzone: ca
Right alloca redzone: cb
Shadow gap: cc
==17925==ABORTING
```

dj0abr, 2021-08-22:
same in bessel.c and many other filters.
I hope it will be fixed soon since it causes random and hard to find crashes,

jgaeddert, 2021-08-23:
Yes, this is critical.