jgaeddert/liquid-dsp #249: [INFO] liquid-dsp is now on conan-center (Windows users will like it)

ruilvo, 2021-08-04:
Hello there!

I took the liberty to make [liquid-dsp available from conan-center](https://conan.io/center/liquid-dsp). Conan is a package manager for C(++). [It took some effort](https://github.com/conan-io/conan-center-index/pull/5956), but it even works transparently with MSVC on Windows! (full disclosure, to get liquid-dsp to compile it gets mingw as a dependency and builds with with gcc, and then uses msvc's tooling to get a link library for dlls). 

So for Windows users this might be a great opportunity to effortlessly use liquid-dsp.

Thanks for the work in the library!

luizrenault, 2021-11-08:
Can you provide brief instructions on how to use this and compile as a dll using msvc?

jgaeddert, 2021-11-16:
This is great! I realize that the non-native support for `float complex` in MSVC is a real pain which is why I've never pursued it. Kudos for the effort here!