jgaeddert/liquid-dsp #250: Complex notch filter

pfeatherstone, 2021-08-19:
Currently ```liquid_firdes_notch()``` creates a real-only filter. So, say you want to drown out frequency ```fc```, you're also doing it for ```-fc```. Example:
```
    const int m = 50;
    const float fc = 0.4;
    vector<float> h_r(2*m+1);
    liquid_firdes_notch(m, fc, 60, &h_r[0]);
```
gives you the following frequency response:
[image]

So I looked at the code, and adapted it for both real and complex filters:
```
    template <typename T>
    struct remove_complex {typedef T type;};
    template <typename T>
    struct remove_complex<std::complex<T> > {typedef T type;};
    template<typename T>
    using remove_complex_t = typename remove_complex<T>::type;

template<typename R>
R kaiser_function(
    size_t i,
    size_t wlen,
    R      beta
)
{
    static_assert(std::is_floating_point<R>::value, "template parameter should be a floating point type");
    if (i >= wlen)
        throw std::runtime_error("i < wlen not satisfied");
    if (beta <= 0)
        throw std::runtime_error("beta should be strictly positive");

    const R t = (R)i - (R)(wlen-1)/2;
    const R r = 2.0*t/(R)(wlen-1);
    const R a = boost::math::cyl_bessel_i(0, beta*sqrt(1-r*r));
    const R b = boost::math::cyl_bessel_i(0, beta);
    return a / b;
}

/*
 * Use template function overloading. 
 * This is a bit less messy than using loads of type traits 
 * to resolve the cases where you want to return complex tone or real tone.
 */
template<typename T>
void tone_(T phase, T& v)
{
    v = std::cos(phase);
}

template<typename T>
void tone_(T phase, std::complex<T>& v)
{
    v = std::polar<T>(T(1), phase);
}

template<typename T>
std::vector<T> notch_filter(size_t m, float fc)
{
    using R = remove_complex_t<T>;
    static_assert(std::is_floating_point<R>::value, "template parameter should be a floating point or complex floating point type");
    
    if (fc < -0.5 || fc > +0.5)
        throw std::runtime_error("fc must be in range (-0.5,+0.5)");
    
    std::vector<T> h(2*m+1);
    R scale = 0;
    
    for (size_t i = 0 ; i < h.size() ; i++)
    {
        const R w     = kaiser_function<R>(i, h.size(), 10.0);
        const R phase = 2*M_PI*fc*i;
        tone_(phase, h[i]);
        scale += std::norm(h[i])*w;
        h[i] *= -w;
    }
    
    for (size_t i = 0 ; i < h.size() ; i++)
        h[i] /= scale;
    
    h[m] += R(1);
    
    return h;
}
```

Now i can create a complex valued filter like so:
```
auto h = notch_filter<complex<float>>(50, 0.4);
```

And i get frequency response:
[image]

IS THIS  A CORRECT FILTER??

It would seem you get the desired behavior but you get funny behavior when ```m``` gets too large. 
So here is the progression:
```
auto h1 = notch_filter<complex<float>>(100, 0.4);
auto h2 = notch_filter<complex<float>>(110, 0.4);
auto h3 = notch_filter<complex<float>>(120, 0.4);
auto h4 = notch_filter<complex<float>>(125, 0.4);
auto h5 = notch_filter<complex<float>>(126, 0.4);
auto h6 = notch_filter<complex<float>>(127, 0.4);
auto h7 = notch_filter<complex<float>>(128, 0.4);
```
These have the following frequency responses:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ANYBODY KNOW WHAT'S GOING ON HERE?

Note that you get similar weird behavior when using ```liquid_firdes_notch()``` for large ```m```, i.e. you get unwanted spikes.

Note that using ```complex<double>``` instead of ```complex<float>``` doesn't solve it. So it doesn't seem to be a floating point precision problem...

pfeatherstone, 2021-08-19:
OK found the bug:
```
const R phase = 2*M_PI*fc*i;
```
->
```
const R phase = 2*M_PI*fc*(R(i) - R(m));
```

Then everything works as expected.

pfeatherstone, 2021-08-19:
So working code is:
```
template <typename T>
struct remove_complex {typedef T type;};
template <typename T>
struct remove_complex<std::complex<T> > {typedef T type;};
template<typename T>
using remove_complex_t = typename remove_complex<T>::type;

template<typename R>
R kaiser_function(
    size_t i,
    size_t wlen,
    R      beta
)
{
    static_assert(std::is_floating_point<R>::value, "template parameter should be a floating point type");
    if (i >= wlen)
        throw std::runtime_error("i < wlen not satisfied");
    if (beta <= 0)
        throw std::runtime_error("beta should be strictly positive");

    const R t = (R)i - (R)(wlen-1)/2;
    const R r = 2.0*t/(R)(wlen-1);
    const R a = boost::math::cyl_bessel_i(0, beta*sqrt(1-r*r));
    const R b = boost::math::cyl_bessel_i(0, beta);
    return a / b;
}

/*
 * Use template function overloading. 
 * This is a bit less messy than using loads of type traits 
 * to resolve the cases where you want to return complex tone or real tone.
 */
template<typename T>
void tone_(T phase, T& v)
{
    v = std::cos(phase);
}

template<typename T>
void tone_(T phase, std::complex<T>& v)
{
    v = std::polar<T>(T(1), phase);
}

template<typename T>
std::vector<T> notch_filter(size_t m, float fc)
{
    using R = remove_complex_t<T>;
    static_assert(std::is_floating_point<R>::value, "template parameter should be a floating point or complex floating point type");
    
    if (fc < -0.5 || fc > +0.5)
        throw std::runtime_error("fc must be in range (-0.5,+0.5)");
    
    std::vector<T> h(2*m+1);
    R scale = 0;
    
    for (size_t i = 0 ; i < h.size() ; i++)
    {
        const R w     = kaiser_function<R>(i, h.size(), 10.0);
        const R phase = 2*M_PI*fc*(R(i) - R(m));
        tone_(phase, h[i]);
        scale += std::norm(h[i])*w;
        h[i] *= -w;
    }
    
    for (size_t i = 0 ; i < h.size() ; i++)
        h[i] /= scale;
    
    h[m] += R(1);
    
    return h;
}
```

pfeatherstone, 2021-08-19:
To give some context: i want to drown out a frequency bin of some signal. Say i have some IQ, I want to zero a 1 of 512 frequency bins for examples. I thought of two ways of doing it:
* convolve with a notch filter and figure out the filter length in order to get the appropriate frequency resolution, i.e. 1/512
* use a polyphase filter bank, using 512 subfilter length, then zero the target frequency bin, then do the reverse transformation.

Anybody have an opinion on which is best? Also, are they mathematically equivalent?

jgaeddert, 2021-08-21:
liquid does support a notch filter off DC: https://github.com/jgaeddert/liquid-dsp/blob/master/include/liquid.h#L2610-L2616

The `firfilt_cccf_notch_example.c` demonstrates this:

[image]

This is the difference between `firfilt_crcf` and `firfilt_cccf`; with real coefficients, the response is always symmetric so a notch at e.g. 0.2 Fs also appears at -0.2 Fs. But with complex coefficients, the notch can be at any specific frequency (and _only_ that frequency) you want.

jgaeddert, 2021-08-21:
As for your question about whether or not a notch filter or channelizer is best, they certainly are not equivalent, but depending upon what you're doing they might give you similar results. Here's a quick run-down of the difference:

  * channelizer approach (e.g .`firpfbch2` analysis followed by `firpfbch2` synthesis) will give you good resolution and the ability to notch frequencies arbitrarily.
  * channelizer will allow you to adjust gains on sub-channels as well, if you would need it
  * the notch filter probably requires a lot less computation
  * the notch filter can operate one sample at a time (one sample in, one sample out); the channelizer would need to work on blocks of samples (e.g. 256 in, 256 out)
  * they would have comparable processing delays (e.g. impulse in would see about the same number of samples delay before appearing on the output)
  * the channelizer would have a small (probably negligible) amount of distortion in the pass-band. It's not a true perfect reconstruction filter, although it's very very close

pfeatherstone, 2021-08-21:
Cheers @jgaeddert for the explanations and pointing out the existing code. Looking at it, you create a DC blocker then mix appropriately. That's equivalent to what I posted the second time round so all good there.

pfeatherstone, 2021-08-21:
I think the channelizer approach is more flexible anyway. You can zero out the exact frequency bins. Using the notch filter, it isn't obvious to me how you choose the filter length to zero out a specified frequency range.

pfeatherstone, 2021-08-21:
I'm sorry, this isn't an issue, more of a discussion.

closed 2021-08-21 by pfeatherstone

jgaeddert, 2021-08-22:
No apology necessary! Nice discussion.