jgaeddert/liquid-dsp #251: possible NCO bug

dj0abr, 2021-08-21:
Hello Joseph,

since about 2 week I had a bug in a mixer application using your library.
It turns out that its caused by the NCO, has it been changed?

The problem is ONLY on 32 bit computers (Raspbian) but all works ok on 64 bit OSs.

When I use the NCO as down mixer and set the "radians per sample" to 0..PI all works fine, but in the range PI..2PI the NCO outputs nothing. Its funny that it is a problem only on 32 bit platforms, but works fine on 64 bit.

Thanks,
Kurt

dj0abr, 2021-08-21:
Hi,

I have fixed it. 
The problem was that the last update changed uint32_t to long, and (signed) long holds only +/- MAXINT/2 instead of MAXINT.

change: 
long retVal = (long)(fpart * 0xffffffff);
to:
uint32_t retVal = (uint32_t)(fpart * 0xffffffff);

now it works on 32 and 64 bit machines

Regards
Kurt

jgaeddert, 2021-09-09:
Sorry for taking such a long time to get back to this. I believe the change happened in fd3f96b1d64 to address some compiler issues. I've effectively reverted that change in b9de495b and confirmed tests pass on a raspberry pi 4.

closed 2021-09-09 by jgaeddert