jgaeddert/liquid-dsp #253: ampmodem requires double precision floating point

knightshrub, 2021-09-11:
I am trying to use liquid-dsp in an embedded system with an ARM Cortex-M4F microcontroller which has hardware floating point acceleration, albeit single precision only (i.e. `float` not `double`).

Unfortunately when using the `ampmodem` the produced binary includes a lot of calls to software double-precision floating point library functions like `__aeabi_dmul`, which are very slow on this particular system and means that I can't use it in real time.

Is there a way to use single precision floating point throughout the library to make use of the available hardware acceleration?

fsheikh, 2021-09-11:
Isn't this a matter of correctly choosing (based on the ARM core under usage) `mfpu` option in the configure [script](https://github.com/jgaeddert/liquid-dsp/blob/master/configure.ac#L220) . At least, this [page](https://www.keil.com/support/man/docs/armclang_ref/armclang_ref_chr1392305424052.htm) lists a couple of compiler options for generating single precision float point numbers

knightshrub, 2021-09-11:
No, this only enables hardware acceleration. However when there are double-precision operations in the code, the compiler will not simple substitute `float` for `double`.

Using the compiler option `-Wdouble-promotion` will throw a lot of warnings for floating point values being promoted to `double` in expressions where the other operand is a `double` or because the compiler treats constants like `1.0` as a `double`.
The compiler option `-fsingle-precision-constant` changes this behavior to interpret `1.0` the same as `1.0f` (both `float`s), but this doesn't seem to fix the issue in case of liquid-dsp.

jgaeddert, 2021-09-11:
@knightshrub this is odd as I try to avoid double-precision operations wherever possible. There are only a few cases where double-precision is needed, but I can't imagine anything to do with the `ampmodem` would need double precision.

knightshrub, 2021-09-12:
@jgaeddert thanks for the reply, I realize my previous comment might have come off as a little rude, I certainly didn't mean it that way.

First of all, this is how I compile libliquid:
```
# remove any leftovers
make distclean
./bootstrap.sh
# this might very well be the culprit, I'm not super familiar with autoconf
# I also had to comment out AC_FUNC_MALLOC and AC_FUNC_REALLOC in configure.ac
CFLAGS='-mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -O3 -fsingle-precision-constant -Wdouble-promotion' LDFLAGS='-specs=nosys.specs' ./configure --host=arm-none-eabi
make
```

It appears that all the references to `__aeabi_d.*` are introduced by the NCO module.
Does the NCO code really require double precision arithmetic?
As an example let's have a look at `nco_crcf_get_freq`:

```asm
080010cc <nco_crcf_get_frequency>:
 80010cc:	f500 5080 	add.w	r0, r0, #4096	; 0x1000
 80010d0:	b538      	push	{r3, r4, r5, lr}
 80010d2:	6880      	ldr	r0, [r0, #8]
 80010d4:	ee07 0a90 	vmov	s15, r0
 80010d8:	eef8 7a67 	vcvt.f32.u32	s15, s15
 80010dc:	ed2d 8b02 	vpush	{d8}
 80010e0:	ee17 0a90 	vmov	r0, s15
 80010e4:	f02d facc 	bl	802e680 <__aeabi_f2d>
 80010e8:	a315      	add	r3, pc, #84	; (adr r3, 8001140 <nco_crcf_get_frequency+0x74>)
 80010ea:	e9d3 2300 	ldrd	r2, r3, [r3]
 80010ee:	f02d fb1f 	bl	802e730 <__aeabi_dmul>
 80010f2:	4b15      	ldr	r3, [pc, #84]	; (8001148 <nco_crcf_get_frequency+0x7c>)
 80010f4:	2200      	movs	r2, #0
 80010f6:	f02d fb1b 	bl	802e730 <__aeabi_dmul>
 80010fa:	f02d fe11 	bl	802ed20 <__aeabi_d2f>
 80010fe:	ee08 0a10 	vmov	s16, r0
 8001102:	f02d fabd 	bl	802e680 <__aeabi_f2d>
 8001106:	a30c      	add	r3, pc, #48	; (adr r3, 8001138 <nco_crcf_get_frequency+0x6c>)
 8001108:	e9d3 2300 	ldrd	r2, r3, [r3]
 800110c:	4604      	mov	r4, r0
 800110e:	460d      	mov	r5, r1
 8001110:	f02d fd9e 	bl	802ec50 <__aeabi_dcmpgt>
 8001114:	b150      	cbz	r0, 800112c <nco_crcf_get_frequency+0x60>
 8001116:	a30a      	add	r3, pc, #40	; (adr r3, 8001140 <nco_crcf_get_frequency+0x74>)
 8001118:	e9d3 2300 	ldrd	r2, r3, [r3]
 800111c:	4620      	mov	r0, r4
 800111e:	4629      	mov	r1, r5
 8001120:	f02d f94e 	bl	802e3c0 <__aeabi_dsub>
 8001124:	f02d fdfc 	bl	802ed20 <__aeabi_d2f>
 8001128:	ee08 0a10 	vmov	s16, r0
 800112c:	eeb0 0a48 	vmov.f32	s0, s16
 8001130:	ecbd 8b02 	vpop	{d8}
 8001134:	bd38      	pop	{r3, r4, r5, pc}
 8001136:	bf00      	nop
 8001138:	54442d18 	.word	0x54442d18
 800113c:	400921fb 	.word	0x400921fb
 8001140:	54442d18 	.word	0x54442d18
 8001144:	401921fb 	.word	0x401921fb
 8001148:	3df00000 	.word	0x3df00000
```

Looking at the source
https://github.com/jgaeddert/liquid-dsp/blob/b9de495b0d529977914a4bbcd1295208e5de07c0/src/nco/src/nco.c#L167-L172
reveals that the code uses `M_PI` which according to the [GNU libm documentation](https://www.gnu.org/software/libc/manual/html_node/Mathematical-Constants.html) is a `double`. This probably leads to the entire expression being promoted to double precision arithmetic,  although the the `-Wdouble-promotion` flag doesn't seem to complain about it.

A piece of code that gcc with `-Wdouble-promotion` does complain about is this:
```
src/nco/src/nco.c: In function 'nco_crcf_mix_down':
src/nco/src/nco.c:283:14: warning: implicit conversion from 'complex float' to 'complex double' to match other operand of binary expression [-Wdouble-promotion]
    283 | *_y = _x * conj(v);
```
[... 72 lines cut ...]
 8001510:	4650      	mov	r0, sl
 8001512:	f02d fc05 	bl	802ed20 <__aeabi_d2f>
 8001516:	4603      	mov	r3, r0
 8001518:	4620      	mov	r0, r4
 800151a:	9c01      	ldr	r4, [sp, #4]
 800151c:	4629      	mov	r1, r5
 800151e:	6023      	str	r3, [r4, #0]
 8001520:	f02d fbfe 	bl	802ed20 <__aeabi_d2f>
 8001524:	4603      	mov	r3, r0
 8001526:	2000      	movs	r0, #0
 8001528:	6063      	str	r3, [r4, #4]
 800152a:	b005      	add	sp, #20
 800152c:	ecbd 8b04 	vpop	{d8-d9}
 8001530:	e8bd 8ff0 	ldmia.w	sp!, {r4, r5, r6, r7, r8, r9, sl, fp, pc}
 8001534:	ec47 6b13 	vmov	d3, r6, r7
 8001538:	eeb0 2a49 	vmov.f32	s4, s18
 800153c:	eef0 2a69 	vmov.f32	s5, s19
 8001540:	ec49 8b10 	vmov	d0, r8, r9
 8001544:	eeb0 1a48 	vmov.f32	s2, s16
 8001548:	eef0 1a68 	vmov.f32	s3, s17
 800154c:	f02d fd78 	bl	802f040 <__muldc3>
 8001550:	ec5b ab10 	vmov	sl, fp, d0
 8001554:	ec55 4b11 	vmov	r4, r5, d1
 8001558:	e7d9      	b.n	800150e <nco_crcf_mix_down+0xae>
```

jgaeddert, 2021-09-12:
Oh, I didn't take it as rude at all! You're right in that `M_PI` is interpreted as a double (difference between `3.14159` and `3.14159f`) because it's a preprocessor macro in the `math.h` header.

Good catch with `conj` vs `conjf` as well. I'll submit a patch and see if that fixes things.

jgaeddert, 2021-09-12:
Made a few changes to the `nco` module. They're temporary until I can figure out a better way to manage types and literals, but hopefully it should work for you.

pfeatherstone, 2021-09-12:
C++ templates ?

knightshrub, 2021-09-12:
@jgaeddert thanks for the temporary changes. I think I found another one in ampmodem.c, should probably be `cargf`.
https://github.com/jgaeddert/liquid-dsp/blob/3d9de34350c7ce3ba130672a1646cf92a927e81d/src/modem/src/ampmodem.c#L301

jgaeddert, 2021-09-13:
Fixed! Thanks!

jgaeddert, 2021-09-13:
I will also mention there is a branch of _liquid_ that includes fixed-point algorithms; however the branch is a bit stale and a dry-run of merging master into it revealed a lot of conflicts. While it would take quite some time to get it up to speed again, it should work much faster on embedded systems, particularly those without floating-point accelerators. Keeping track of bit precision can be a pain, of course, but if low processing footprint is your goal, fixed-point is the way to go.