jgaeddert/liquid-dsp #257: PLL : keep track of dphi

pfeatherstone, 2021-09-24:
So I've played with the different ways of doing PLL in liquid.
There are two methods I can see:
- use NCO object which uses a very simple model
- use method described in https://liquidsdr.org/doc/tutorial-pll/ which uses an active lag design.

Now the first method is great since it's quite simple and it estimates both ```phi``` and ```dphi```. So once you've locked onto a carrier, you can adjust frequency as well as phase.

The second seems to be way more accurate but can only track phase. So you can't really lock onto a signal, since when you no longer have any "truth" data, you won't be able to adjust frequency.

Is there a way to enhance the active lag design to also keep track of frequency/dphi?

Cheers

pfeatherstone, 2021-09-24:
Do you have to combine both into a single loop?
Maybe use the Costas loop as your main algorithm but use the other algorithm to refine phase when you have truth data?

pfeatherstone, 2021-09-25:
I'm not an electrical engineer so what I'm about to suggest could be naive. It seems that all the PLL is doing is smoothing phase error over time using a filter, in both cases above, using an IIR filter. Why can't we do the same with an averaging filter to estimate phase error and difference filter to estimate frequency error? How about that?

pfeatherstone, 2021-09-27:
@jgaeddert any thoughts ?

jgaeddert, 2021-09-27:
It's not a naive thought that the PLL is smoothing the phase error over time. The important piece is balancing the filter gain to keep the loop in its stable region. You can easily extend this to higher order by incorporating frequency offset estimates. The `nco` object in _liquid_ actually already does this.

jgaeddert, 2021-09-27:
See the example here: https://liquidsdr.org/blog/pll-simple-howto/

pfeatherstone, 2021-09-27:
Yeah so that's what I was using but it takes a lot of iterations for it to converge well. Will any kind of smoothing do ?

jgaeddert, 2021-09-27:
I think you'll find you can't set the values arbitrarily otherwise the loop won't be stable; you need to balance between the phase and frequency gains. You can increase the loop bandwidth to reduce convergence time. Honestly a Kalman filter is probably near-optimal, but without implementing it I'm guessing it would have similar performance.

Feel free to play around with the numbers and see if you can find something that works better.

pfeatherstone, 2021-09-27:
Are Kalman filters used in PLLs ? I love the sound of that

jgaeddert, 2021-09-28:
They certainly can be, and I'm sure it's been done before. It's non-linear, of course, so you'd have to use an extended Kalman filter at the least.

pfeatherstone, 2021-09-28:
There are loads of papers on this but no open source implementation as far as i can see.

pfeatherstone, 2021-09-28:
This would be an awesome feature in liquid, even if it was for demo purposes

pfeatherstone, 2022-02-25:
https://hal.archives-ouvertes.fr/hal-03203769/document

jgaeddert, 2022-02-26:
Great reference! And should work very well for non-linear frequency offsets as well as timing loop control. I've worked on extended Kalman filters before, and have been interested in applying it to tracking loops.

Also, this article is one of the rare exceptions to [Bettridge's Law](https://en.wikipedia.org/wiki/Betteridge's_law_of_headlines)

noonafter, 2024-11-26:
In NCO object, NCO(_pll_step) method adjust frequency first, then adjust phase. However, **from the perspective of sequential circuits**, the frequency and phase should be adjust synchronously, which means when phase is adjust, the frequency remains the same as the previous value. So mehtod NCO(_adjust_phase) should be called first, then NCO(_step), and finally NCO(_adjust_frequency). If we do this, NCO(_step) won't be called again, just like the example in  #https://liquidsdr.org/doc/nco/#section-nco-pll-interface.

noonafter, 2024-11-26:
In NCO object, NCO(_pll_step) method adjust frequency first, then adjust phase. However, from the perspective of sequential circuits, the frequency and phase should be adjust simultaneously, which means when phase is adjust, the frequency remains the same as the previous value. So mehtod NCO(_adjust_phase) should be called first, then NCO(_step), and finally NCO(_adjust_frequency). If we do this, NCO(_step) won't be called again, as the example in https://liquidsdr.org/doc/nco/#section-nco-pll-interface.