jgaeddert/liquid-dsp #258: undefined reference to `volk_32fc_x2_dot_prod_32fc'

solomonbstoner, 2021-09-26:
I successfully ran `make` and `sudo make install`, but running `make check` and `make examples` give me the following error. I installed volk successfully before installing liquid-dsp. 

```
/usr/bin/ld: libliquid.a(dotprod_crcf.mmx.o): in function `dotprod_crcf_execute':
/home/solomonbstoner/Documents/GitClones/liquid-dsp/src/dotprod/src/dotprod_crcf.mmx.c:176: undefined reference to `volk_32fc_x2_dot_prod_32fc'
collect2: error: ld returned 1 exit status
make: *** [makefile:1322: xautotest] Error 1
```

solomonbstoner, 2021-09-26:
Deleted liquid-dsp and redownloaded it using `git clone`. Installation works now

closed 2021-09-26 by solomonbstoner

jgaeddert, 2021-09-26:
That's quite odd. Liquid doesn't (and hasn't ever) used volk for any of its processing. Not sure why it would try to link to it.

solomonbstoner, 2021-09-28:
Oops, my bad. It was the modified version of liquiddsp for another software for radio reslience competition. i mixed up the original and modified liquiddsp. This completely slipped my mind

jgaeddert, 2021-09-28:
Oh no worries! I was just quite confused by the error. Good luck!