jgaeddert/liquid-dsp #259: Question : bug in modem_common.c ?

pfeatherstone, 2021-10-01:
I've spotted something weird.
https://github.com/jgaeddert/liquid-dsp/blob/ee10d0d8a4a03404d00c08a17e5a8550321aa9f1/src/modem/src/modem_common.c#L522

Shouldn't that be:
```
return cargf(_q->x_hat*conjf(_q->));

```

pfeatherstone, 2021-10-01:
Give or take a sign-convention, maybe this is ok since for small ```phi```, ```phi ~ sin(phi)```.
Or have I miss-understood something?

jgaeddert, 2021-10-02:
Ideally yes but taking the `cargf ` is a lot slower than taking `cimagf`. Taking just the imaginary component is usually sufficient for PLL tracking.

pfeatherstone, 2021-10-02:
Hmm but if we're aiming for correctness, surely it's best to take the phase right ?

jgaeddert, 2021-10-02:
I suspect there's actually very little difference in practice but it would be an interesting experiment to see!

pfeatherstone, 2021-10-02:
Ok cheers