jgaeddert/liquid-dsp #265: Question: couldn't we make a decision-directed version of symsync?

pfeatherstone, 2021-11-08:
First of all, why is the following line:
https://github.com/jgaeddert/liquid-dsp/blob/471f46660ddfab7e6713341324f473b7785da101/src/filter/src/symsync.c#L521
not:
```
_q->q = tanh( _mf*_dmf );
```

?

(I'm about to purchase Mengali so i can read the paragraph you reference (£250 !))

Then, couldn't we have a decision directed version of ```symsync``` where the timing error uses the true symbol value rather than the output of the matched filter?
So:
```
e = tanh(y_true, y_dot)
```
instead of 
```
e = tanh(y, y_dot)
```

pfeatherstone, 2021-11-08:
Then all of the components of ```symtrack``` could be decision-directed too! Indeed this value
https://github.com/jgaeddert/liquid-dsp/blob/471f46660ddfab7e6713341324f473b7785da101/src/framing/src/symtrack.c#L358-L359

could be used to update the symsync object.

pfeatherstone, 2021-11-09:
In fact, couldn't you just cross-correlate with ```sym_out``` to get the time delay (timing phase or whatever it's called)

pfeatherstone, 2021-11-09:
Side question: is there a difference between what ```symsync``` does and a delay locked loop (DLL)?

jgaeddert, 2021-11-16:
So one nice thing about what's done in `symsync` is that you can be completely agnostic to the modulation format. Sure, knowing the modulation scheme gives you a performance boost, but in most situations this is sufficient.

The `symtrack` class has the ability to operate without knowledge of the specific modulation scheme as well, actually. So you can recover timing, the bulk of the carrier offset, and even equalization even blind. Seemed like a useful design choice. Of course having knowledge of the modulation scheme helps you out.

Yeah, Mengali is expensive. If you are affiliated with a university, you might find a copy in its library and not have to buy it.

jgaeddert, 2021-11-16:
`symsync` is a type of delay-locked loop. In general, this kind of flow control can be tricky, and stability can be an issue. I'm sure there are better ways to do this (e.g. extended Kalman filters) but this seemed like a reasonable balance between performance and computational complexity.

pfeatherstone, 2021-11-16:
It would be really interesting to see a Kalman filter based timing recovery algorithm. Do you know of a book that elaborates on this?

jgaeddert, 2021-11-23:
Not specifically. Kalman filters can be fairly computationally complex which is one of the reasons I've avoided them (requires matrix inversion each step). Also, while the regular (linear) Kalman filter is optimal, the extended Kalman filter for non-linear problems is suboptimal so typically PID controllers work pretty well for this. But when you're dealing with symbol timing recovery simultaneously with carrier phase and equalization, things get trickier.