jgaeddert/liquid-dsp #273: modem API overhaul breaks API backward compatibility

szpajder, 2021-12-12:
Hello Jonathan,

Thank you for your work on this great project.

I've just noticed that in the master branch of liquid-dsp `modem` data type and `modem_*` functions have been renamed to `modemcf`. This is an API breakage - apps depending on `modem` will fail to build. My question is: will the next version of liquid-dsp be a major release with a library having a different soname (say, liquid-2) and include files installed in a separate directory (say, `/usr/local/include/liquid-2`), so that two incompatible versions could be installed simultaneously? If not - can you please keep the old `modem` and `modem_*` symbols to avoid the breakage? Otherwise it's going to be quite hard to have apps using the old API and apps using the new API installed at the same time.

jgaeddert, 2021-12-12:
Fair point. Having more frequent releases to delineate API changes, etc. is probably helpful as well. Thanks for bringing this up.

szpajder, 2021-12-12:
It's not really about the frequency of releases. You do them when you get a particular set of features ready for a production use, be it once every month or quarter or year, it doesn't really matter. My point is that it's OK to make API changes that don't break compatbility with prior releases (eg. new functions added), but breaking changes (functions removed or renamed, parameters changed, etc) shall be as infrequent as possible and communicated clearly in the changelog and preferably also with [semantic versioning](https://semver.org/). [Here](https://web.archive.org/web/20201109040438/https://plan99.net/~mike/writing-shared-libraries.html) is a good summary of dos/donts related to shared libraries. Doing otherwise makes the lib difficult to work with for distribution maintainers and developers who eventually may just come up to the conclusion that the only way to use it reliably is to bundle the library code in the source tree of their project, which is a terrible solution. It's also a pain for the end users who want to install multiple apps which use different incompatible versions of the library. There can only be one libliquid.so in the system, so in order to have several versions, one has to resort to static builds or fiddling with multiple installation dirs and LD_LIBRARY_PATH, which is very messy.

jgaeddert, 2021-12-12:
That's fair, and you're right in that I haven't been adhering to semver conventions. I'm sorry this caused some frustration for you, but I do appreciate you bringing it up to me (and being quite civil about it!).

In the interim I can revert the `modem` changes, but at a certain point a lot of the APIs need to be changed to be made consistent. My original thinking behind messing with `master` is that it's bleeding edge, so YMMV, but I suppose this isn't the best approach when the version doesn't change.

szpajder, 2021-12-12:
Messing with `master` is not a problem - in fact, many projects are developed by commiting bleeding edge stuff to `master`. Given that liquid-dsp is packaged in many distributions, I advise users of my apps to install the packaged version, if available. It's just one user (or two) who decided to go with the latest code from `master` and this is how I got to know about the recent API change. So right now the breakage is not a huge deal - until it's released as a stable version which eventually will go into the distros.

If you plan for renaming even more functions, then reverting the `modem` changes would probably go against these plans, right? In this case it would be good enough to leave `modemcf_*` in place and provide backward-compatibility `modem` and `modem_*` macros until the next major release with a new soname is out.

jgaeddert, 2021-12-22:
Made a change with ed8bb281eceb206c1f97c4007071fb8cd2518733 that I think (temporarily) addresses this. Let me know if this works for now.

szpajder, 2021-12-22:
Thank you.

It works partially, ie. the program that utilizes `modem_*` calls now builds and runs correctly. However if the program has been built with liquid-dsp 1.3.2 and the library gets upgraded to the current master, then that binary no longer runs:

```
./dumphfdl_built_with_liquid_1.3.2: symbol lookup error: ./dumphfdl_built_with_liquid_1.3.2: undefined symbol: modem_create
```

Apologies for suggesting that macros would be enough to deal with this problem. They only fix the API incompatibility, but the ABI is still incompatible, because the new lib is missing symbols which were present before. The only way to deal with this is to provide a full set of wrapping functions, like:

```C
modem modem_create(modulation_scheme scheme) {
    return modemcf_create(scheme);
}
```

And so forth.

Again, I don't mind too much about the bleeding edge code in master being incompatible with the last stable release. I'm just concerned about the next stable release being incompatible with previous releases. Distributions pick new versions at different paces - for example Arch Linux adopts the latest version quickly, while Debian/Ubuntu/RedHat have their rigid release cycles which cause older versions of libraries and apps to be kept around in packages for a significant amount of time. If this happens, then it's going to be really tough to write software which works under various distributions that ship different versions of liquid-dsp.

jgaeddert, 2021-12-23:
Good point. Try 4b84350a87b11c6a94b70767e42916634d4f2c7d

jgaeddert, 2021-12-23:
Consistent releases is something I've (clearly) struggled with. My biggest hangup is determining what's an appropriate change to provide in a release. I think I'm letting the perfect be the enemy of the good.

szpajder, 2021-12-23:
This one works great. Old binaries still run and the apps using `modem_*` compile without changes.

Thank you!

closed 2021-12-23 by szpajder