jgaeddert/liquid-dsp #275: msresamp buffer size issue

steffenb7333, 2022-02-16:
According to the example in https://liquidsdr.org/doc/msresamp/ the size of the output buffer can be estimated. It must be an upper bound, else memory corruption may occur. Sometimes this assumption does not hold and the resampler does return more data than expected.

The following test-case is one example (c++17, catch2):
```
TEST_CASE("estimate is an upper bound")
{
	constexpr auto src = 100.f;
	constexpr auto dst = 201.f;

	constexpr auto sz = 1000;

	constexpr auto rate = dst/src;
	const auto estimate = static_cast<std::size_t>(::ceilf(sz * rate));
	auto source = std::vector<std::complex<float>>(sz);
	auto buffer = std::vector<std::complex<float>>(estimate+2); // +2 is the error, else leads to "double free or corruption (out)"

	msresamp_crcf q = msresamp_crcf_create(rate, 60.0f;);
	unsigned int num_written = 0; // number of values written to buffer
	msresamp_crcf_execute(q, source.data(), source.size(), buffer.data(), &num_written);
	msresamp_crcf_destroy(q);

	REQUIRE(estimate >= num_written);
	// fail: 2010 vs 2012
}
```

jgaeddert, 2022-02-17:
That's curious. I should definitely write a test for that. Thanks for bringing this to my attention!

In general, buffer allocation (or decent interface design) is difficult for resamplers. I might change the interface to be easier in version 2

jgaeddert, 2022-02-27:
I've added methods for `resamp` and `msresamp` objects to get the number of output samples for a given input length in 0d08726c3b9c01ab273c841672bf3d2e1c404d17.

```c
msresamp_crcf resamp = msresamp_crcf_create(123.45678f, 60.0f);

unsigned int num_output = msresamp_crcf_get_num_output(resamp, 1024);

msresamp_crcf_destroy(resamp);
```

Note that the number of outputs can change based on the internal state of the object. But I've written a set of tests to ensure that the number that's returned by this method matches what it actually computes.

closed 2022-02-27 by jgaeddert