jgaeddert/liquid-dsp #276: autotest eqlms_08 fails reproducibly with one seed

andreasbombe, 2022-03-03:
The test eqlms_08 fails reproducibly for me with the seed 1645848426. Other seeds work, this is so far the only fail I have seen.

If the seed isn't sufficient to reproduce this issue I will gather some more data.

jgaeddert, 2022-03-03:
I am able to reproduce this on my end when running with 0d08726c3b9c01ab273c841672bf3d2e1c404d17 on an x86_64 Linux box:

```shell
./xautotest -R 1645848426 -x

...
eqlms_08:
rmse : -18.504 dB
  TEST FAILED: src/equalization/tests/eqlms_cccf_autotest.c line 118 : expected rmse (-1.85E+01) < -20.0f (-2.00E+01)
...
```

jgaeddert, 2022-03-03:
Here's the distribution of the first 8,000 seeds:

[image]

Apparently 42/8,000 fail with a -20 dB threshold (about 0.5 %)

jgaeddert, 2022-03-03:
I'll have to think of a test that's more consistent. Perhaps running with more symbols.

jgaeddert, 2022-03-06:
Should be fixed in 2e9d1316

jgaeddert, 2022-03-06:
The problem was that the learning rate was set too high; dropping the learning rate down enables more consistent convergence. Here's how the above plot looks when I drop the learning rate down from 0.7 to 0.3:

[image]

closed 2022-03-06 by jgaeddert