jgaeddert/liquid-dsp #278: info interpolator

AntonioRiccelli77, 2022-04-12:
Hello guys
I would need an application that acts as an intermediary such as the firinterp_crcf_example with my application in a pipe so that the samples captured by the adc of a sound card with sr at 192000khz can be transposed to 2048000 of my unsigned 8-bit IQ modulator , can you give me a tip?
Thanks

jgaeddert, 2022-04-14:
Well, since 2048 kHz and 192 kHz aren't integer multiples of one another, you'll need to use a resampler. Probably the easiest one would be the rational rate resampler (rresamp_crcf) since 2048 = 192 * 32/3.

```c
rresamp_crcf q = rresamp_crcf_create_default(32,3);

float complex buf_in[32];
float complex buf_out[3];

// repeat as needed
{
    rresamp_crcf_execute(q, buf_in, buf_out);
}

rresamp_crcf_destroy(q);
```

Here's a [full example](https://github.com/jgaeddert/liquid-dsp/blob/master/examples/rresamp_crcf_example.c)

The only thing to remember is that your input and output types are expected to be complex, so if you need to modulate your sound card output from a real-valued signal with e.g. frequency modulation, you'll want to use a [modulator](https://github.com/jgaeddert/liquid-dsp/blob/master/examples/freqmodem_example.c)

AntonioRiccelli77, 2022-04-15:
Hello
Ty for your support, I tried to create an example but unfortunately my standard knowledge of C doesn't allow me to create a real working example, I would need to treat with modulator and rational resampler trough a FIFO file, would you be kind enough to post me a complete example with this approach?
Thanks in advance

closed 2022-08-25 by jgaeddert