jgaeddert/liquid-dsp #280: State of fskframegen and fskframesync?

glv2, 2022-05-02:
Hi,

I was thinking of using ```fskframegen``` and ```fskframesync``` in a project (and adding some missing functions like ```fskframegen_create_set```, ```fskframesync_create_set``` or ```fskframesync_is_frame_open```).

However when testing them I saw that they output a lot of warnings and debug messages.
Are they working, or are they in development?

jgaeddert, 2022-05-08:
The `fskframegen` and `fskframesync` are still a work in progress.

brian-armstrong, 2022-05-08:
If you want to try another attempt at FSK framegen/framesync, I have a working version here https://github.com/jgaeddert/liquid-dsp/pull/159

campagn1, 2022-12-15:
Dear all, I was wondering if there is any news on `fskframegen` and `fskframesync` and if https://github.com/jgaeddert/liquid-dsp/pull/159 is going to be merged in master at one point. 

Best regards, 

Filippo

seonlee2013, 2022-12-15:
好啊，邮件收到啦！
Hi, your email is arrived safely

campagn1, 2023-05-28:
Dear all, I was wondering if there are any updates on this.
Best regards,
Filippo

jgaeddert, 2023-06-01:
Sorry. I'll take a look at this again.

jgaeddert, 2023-06-02:
The challenge is that the referenced branch has diverged quite a bit from master. I can merge it in, but I will need to change the interface to be more consistent.