jgaeddert/liquid-dsp #284: Error attempting "make examples"

AndresChoza, 2022-05-16:
Hi, I've already run the "make" command as I was having trouble before but now I cannot run the "make examples" command supposedly because of some undefined references to functions contained in "math.h":

[image]

Any tips?

jgaeddert, 2022-05-16:
You're seeing a linker error. It's not finding the math library. I suspect you have it installed but it isn't in your path. What do you get when you run the following?

    gcc --print-file-name=libm.a

AndresChoza, 2022-05-18:
Hi there.

I get the following:

> choza@chozaUbuntu:~/Documents/liquid-dsp$ gcc --print-file-name=libm.a
/usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libm.a

I tried to modify the makefile so that the "-lm" linker flag was specifically at the end like this:

[image]

Now I don't get the error reported above, but instead I'm getting this while trying out "make examples": 

[image]

AndresChoza, 2022-05-18:
Also, since we're on the same "trouble with make" page, I'm having trouble building the docs with "make doc".

[image]

Could these be related to each other?

jgaeddert, 2022-05-20:
1. Your modifications to the makefile aren't really correct (they imply that `-lm` is a dependency on the objects), but it seems as though `make` throws together the compiler/linker commands in the right order to get this to work.
  1. The undefined references to the `fec` library are a result from its interface having changed. If you can live without convolutional & Reed Solomon codes, you might try to compile without them. I can help with this.
  1. This is questionable. I haven't bundled documentation with liquid since v1.2.0 (about 10 years ago)

AndresChoza, 2022-05-20:
Right, so what can I do to fix the 'make doc' command? I'm in need of this documentation

Thanks beforehand.

jgaeddert, 2022-06-12:
The documentation is not part of the main repository, but is built in https://liquidsdr.org

closed 2022-08-25 by jgaeddert