jgaeddert/liquid-dsp #287: How do I build a script using this library using GCC?

geajack, 2022-06-08:
I'm on Ubuntu 20.04. I cloned this repo and `make install`ed version 1.4.0. Then I ran `sudo ldconfig` When I try to build a program that uses liquid, I get "undefined reference" errors. I tried setting `LD_LIBRARY_PATH=/usr/local` but no dice.  Tried variants on the command like `gcc -o test test.c`, `gcc -o test -lliquid test.c` etc, but I can't figure it out. How should I link the libraries?

jgaeddert, 2022-06-09:
Depending upon the version of GCC you're using you might need to put the linker library flags at the end of the command line. Try:

gcc -o test test.c -lliquid

geajack, 2022-06-09:
Yes, that was all it was in the end. Also I think I forgot to actually export the library parth, not sure if that mattered. Sorry about that.

closed 2022-06-09 by geajack

jgaeddert, 2022-06-09:
No worries!