jgaeddert/liquid-dsp #289: Emscripten build fails with libtool error

shamrin, 2022-06-26:
Emscripten WebAssembly build fails with `libtool` error:

```
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/libliquid.o is not an object file (not allowed in a library)
...
```

Even though `libliquid.o` seems to be fine:

```
~/s/liquid-dsp (master|…) [1]$ file src/libliquid.o
src/libliquid.o: WebAssembly (wasm) binary module version 0x1 (MVP)
```

I've followed Emscripten-recommended steps:

```
./bootstrap.sh
emconfigure ./configure
emmake make VERBOSE=1
```

<details>
  <summary>click here to see full build logs</summary>

```
~/s/liquid-dsp (master|✔) $ ./bootstrap.sh
~/s/liquid-dsp (master|…) $ emconfigure ./configure
configure: ./configure
checking for gcc... /Users/shamrin/src/emsdk/upstream/emscripten/emcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether /Users/shamrin/src/emsdk/upstream/emscripten/emcc accepts -g... yes
checking for /Users/shamrin/src/emsdk/upstream/emscripten/emcc option to enable C11 features... none needed
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for a BSD-compatible install... /usr/bin/install -c
checking for ranlib... /Users/shamrin/src/emsdk/upstream/emscripten/emranlib
checking for error_at_line... no
checking build system type... x86_64-apple-darwin19.6.0
checking host system type... x86_64-apple-darwin19.6.0
checking for GNU libc compatible malloc... yes
checking for GNU libc compatible realloc... yes
checking for main in -lc... yes
checking for main in -lm... yes
checking for malloc... yes
checking for realloc... yes
checking for free... yes
checking for memset... yes
checking for memmove... yes
checking for sinf... yes
checking for cosf... yes
checking for expf... yes
checking for cargf... yes
checking for cexpf... yes
checking for crealf... yes
checking for cimagf... yes
checking for sqrtf... yes
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
[... 379 lines cut ...]
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/random/src/randweib.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/random/src/randgamma.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/random/src/randnakm.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/random/src/randricek.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/random/src/scramble.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/sequence/src/bsequence.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/sequence/src/msequence.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/utility/src/bshift_array.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/utility/src/byte_utilities.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/utility/src/msb_index.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/utility/src/pack_bytes.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/utility/src/shift_array.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/utility/src/utility.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorf_add.port.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorf_norm.port.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorf_mul.port.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorf_trig.port.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorcf_add.port.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorcf_norm.port.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorcf_mul.port.o is not an object file (not allowed in a library)
error: /Library/Developer/CommandLineTools/usr/bin/libtool: file: src/vector/src/vectorcf_trig.port.o is not an object file (not allowed in a library)
make: *** [libliquid.ar] Error 1

```
</details>

seonlee2013, 2022-06-26:
好啊，邮件收到啦！
Hi, your email is arrived safely

shamrin, 2022-06-26:
Follow-up. I've had a bit more luck with `emconfigure ./configure --host=wasm32`:

```
...
/Users/shamrin/src/emsdk/upstream/bin/llvm-ar: warning: creating libliquid.a
/Users/shamrin/src/emsdk/upstream/emscripten/emranlib libliquid.a
/Users/shamrin/src/emsdk/upstream/emscripten/emcc -g -O2    -Wall -fPIC -Wno-deprecated -Wno-deprecated-declarations  -shared -Xlinker -soname=libliquid.so -o libliquid.so -Wl,-whole-archive libliquid.a -Wl,-no-whole-archive -lm -lc 
emcc: warning: linking a library with `-shared` will emit a static object file.  This is a form of emulation to support existing build systems.  If you want to build a runtime shared library use the SIDE_MODULE setting. [-Wemcc]
emcc: warning: ignoring unsupported linker flag: `-soname=libliquid.so` [-Wlinkflags]
```

Let's see if I can link against this version of liquid...

shamrin, 2022-06-26:
It works just fine! I've compiled the test program with `emcc test.c libliquid.a -o output.html -sSINGLE_FILE`

shamrin, 2022-06-26:
@jgaeddert Should I close this ticket? If you want support WebAssembly directly, I can make a PR to add instructions for Emscripten.

jgaeddert, 2022-08-24:
@shamrin a PR would be useful! Thanks